prusa3d/PrusaSlicer #5402: High CPU usage when idle

tg73, 2020-12-08:
### Version
2.3.0-beta2 win x64

### Operating system type + version
Win 10

### Behavior
In the last 24 hours or so I noticed that my PC was running its cooling fans at high speed when the system was idle (ie, screen off, not in use). I've used SysInternals procmon profiling events to monitor usage (handy when you need to monitor when the system idle). I see two out of the three PS processes I have running each using near 12.5% of total CPU, which is equivalent to each process maxing out one of my 8 CPU cores. These instances of PS are not busy slicing or anything - they are "just sitting there". I also see that these PS process use around 3% CPU each even when the window is minimized and PS is "doing nothing". I will continue to investigate and update in due course, but wanted to flag this issue now and I'm interested to know if anyone else is seeing this. Note that my attention was drawn to the problem by my PC running loudly, and I did not see this with previous versions of PS despite often having several PS processes sitting around.

bubnikv, 2020-12-08:
> I see two out of the three PS processes I have running each using near 12.5% of total CPU

I cannot reproduce with the current master.

We have an issue with the notifications fading out, that is being worked on. In beta2, if there is a notification shown which will eventually fade out, the render loop is running at a full throttle.

tg73, 2020-12-08:
I think the notification issue is the culprit. Of the three PS instances, the two burning CPU have post-export-gcode notifications, the one not burning CPU does not have a notification. The notifications have been there for hours, I don't think they ever fade out:

[image]

[image]

When I close the notification, the CPU usage drops to zero-ish.

I wonder if the maxed-out CPU usage when the PC is idle - and notably when idle enough that screen has gone to sleep - is caused by invalidated DirectX surfaces or contexts or similar concept (or opengl equivalent) causing some even tighter loop?

scottmudge, 2020-12-08:
It's wglMakeCurrent(), which is consuming a vast majority of the CPU. It makes sense, if there's a lot of context switching. wglMakeCurrent() is not asynchronous and is blocking, so it will appear as high CPU usage until the OpenGL context yields to PS.

[image]

I will refactor and look into fixing the issue.

bubnikv, 2020-12-08:
> I will refactor and look into fixing the issue.

Don't waste your time. If it is the notification issue, we are working on
it.

scottmudge, 2020-12-08:
ok, let me know the solution. OpenGL context juggling is annoying to deal with. I never found a good asynchronous way to do it.

scottmudge, 2020-12-08:
It looks like it's ` _picking_pass()`  inside `void GLCanvas3D::render()`.

Specifically glReadPixels(), which is not very efficient. It's enabled all the time for some reason. Commenting this out brings the CPU back down to sub 1%.

kocikdav, 2020-12-09:
The issue of notifications burning lot of CPU/GPU is hopefully fixed at 489c7504519f9948887c8edc4b4ec5ed1bb69bf1.

bubnikv, 2020-12-12:
https://github.com/prusa3d/PrusaSlicer/commit/489c7504519f9948887c8edc4b4ec5ed1bb69bf1 significantly lowered GPU consumption when a notification is displayed in the 3D scene, which is fading out or which will fade out. We still did not solved the CPU issue, where one thread is consuming 50% of a single core.

lukasmatena, 2020-12-18:
This should be fixed in [2.3.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc1), including the problem with 50% CPU usage. Could you please retest and close the issue if happy? Thanks.

tg73, 2020-12-28:
@lukasmatena it was looking promising, but I've just had the experience of returning to a noisy-fan screen-off idle PC. I will do more tests tomorrow to determine what the cause is (it might not even be PS).

tg73, 2020-12-31:
@lukasmatena Appears to be fixed in 2.3.0-rc1, have not been able to reproduce with any notifications or other text panels in the 3D view. Closing.

closed 2020-12-31 by tg73