prusa3d/PrusaSlicer #5405: Things that don't fit in existing profiles.

towlerg, 2020-12-08:
### Version
All

### Operating system type + version
Feature request

### 3D printer brand / version + firmware version (if known)
Feature request

### Behavior
Feature request

_Is this a new feature request?_

Yes. Ref this thread https://forum.prusaprinters.org/forum/prusaslicer/things-that-dont-fit-in-existing-profiles/

In  particular -
"The more I think about it, the more I like the idea of a "job" or "project" profile that contains over-rides for any other (print, printer or filament) setting on a job-specific basis. Today, I'm using THIS printer with THIS filament with THIS set of print settings but I want to print something delicate. Tomorrow, I'm using the same printer with the same filament with the same basic print settings, but I'm printing something chonky."

nordurljosahvida, 2020-12-08:
The part profile is an interesting idea.

bubnikv, 2020-12-11:
Couldn't you import / export config?
You may even import the whole config from a G-code.

towlerg, 2020-12-11:
I'm looking at it the other way round, not from a part already sliced but a now part. For example I constantly forget to set seam position. If there was a Part setting.

If it's going to be done, I'd plan for 3 sub tabs but I suspect that apart from fan speed, the only overrides would be in Print settings. You could also throw in an optional Notes item.

There is already a precedent for this optional override thing, overrides in Filament Setting, same idea but per part instead of per filament.

I understand it's a lot of work but it will differentiate PrusaSlicer even more. 

Thanks for considering, hope it pans out.

George

closed 2026-09-29 by prusaslicer-bot[bot]