prusa3d/PrusaSlicer #5407: [2.3.0-beta2] Odd bad slicing related to infill (?) percentage

MattBlissett, 2020-12-08:
### Version
PrusaSlicer-2.3.0-beta2+linux-x64-202012051122.AppImage

### Operating system type + version
Kubuntu focal

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk3S, stock, default firmware

### Behavior

Something very strange happens in beta2 (left), with the resulting G-code making a mess compared to G-code generated by beta1 (right):

[image]

Steps to reproduce:
1. (This happens with a fresh configuration of Prusa Slicer, as well as with my existing -- unupdated -- configuration I was using with beta1.)
1. Load the attached project into beta2
2. Slice it.  I see a print time of 41 minutes.
3. Set the infill percentage to 15%
4. Slice it. I see a print time of 42 minutes.
5. Set the infill percentage back to 0%.
6. Slice it. I now see a print time of 1h5m, and the G-code makes the mess on the left.  The model on the right is printed with the same filament, after rolling back my configuration settings and slicing using beta1.

[Flexi-Rex-Baby-issue.zip](https://github.com/prusa3d/PrusaSlicer/files/5659596/Flexi-Rex-Baby-issue.zip)

bubnikv, 2020-12-10:
@rtyr looked into this issue.

According to him, it seems to be an update issue if a parameter is changed for an object part after slicing and resliced. According to @rtyr it happens even in PrusaSlicer  2.2.0. We will not be able to fix it for the PrusaSlicer 2.3.0 release.

For example, @rtyr was able to reproduce the issue, if he changed the infill of a single object part to 30%, resliced, then switched to 0% and resliced again. After reslicing, the perimeters are mangled. Not only the preview is mangled, but the G-code as well.

[image]

bubnikv, 2020-12-10:
For my reference. Most likely Layer::merge_slices() is guilty.

bubnikv, 2020-12-11:
This issue should be fixed with a2959ec944c8a1423a382169818284d500270f15
    This is a fix of a bug, which was in Slic3r forever, where raw slices
    were not cached, but recalculated from classified regions, where
    merging the regions did not produce the original contour reliably.

closed 2020-12-11 by bubnikv