prusa3d/PrusaSlicer #5408: Deadlock when using only one processor core

supermerill, 2020-12-08:
### Version
2.3.0-beta2

### Operating system type + version
windows 10 (up-to-date)

### Behavior
On windows, i set the affinity for prusaslicer to only one core.
then i add a cube and slice it.
Then it's deadlocked on Thread.cpp:l225, because it's only using one thread at a time and the first one wait the next, but it can't spawn.

To avoid commenting the whole thing, i changed the wait by a wait_until and commented the asserts.

bubnikv, 2020-12-08:
> On windows, i set the affinity for prusaslicer to only one core.

Why are you doing that?

> To avoid commenting the whole thing, i changed the wait by a wait_until and commented the asserts.

I have no idea what you are trying to achieve and why.

supermerill, 2020-12-08:
I'm trying to watch a video while slicing a big file.

bubnikv, 2020-12-08:
name_tbb_thread_pool_threads()
nthreads = 1;

bubnikv, 2020-12-08:
advice has been given.
If you need your SS to work with thread affinity forced from the outside, you are on your own.

closed 2020-12-08 by bubnikv

supermerill, 2020-12-12:
Note that i had multiple deadlock, where the BgSlcPcs is waiting 
in result.wait();
in background_process.set_thumbnail_cb
in Plater::priv::priv(Plater *q, MainFrame *main_frame)

with no other thread working, and the gui completely freezed.
Also i didn't messed with thread affinity this time.