prusa3d/PrusaSlicer #5409: [Feature request] Ironing whole layers

kusiak, 2020-12-08:
### Version
2.3.0 Beta 3

### 3D printer brand / version + firmware version (if known)

_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
When ironing top surfaces slicer ignores areas that will not be visible
this is a problem as on some models this induces a lot of retractions, movements and blobs left on top surface.
* _Steps needed to reproduce the problem_
Take any model with lots of symbols on top surface
* _Expected Results_
Provide an option in the slicer settings for ironing the whole top surface, at once, then print next layers on top of the smoothed surface
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]
Nozzle movements for every little inside space
[image]

_Is this a new feature request?_
yes

#### Project File (.3MF) where problem occurs
[example_issue_file.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5661303/example_issue_file.3mf.zip)

I'm happy to provide more context if needed

stepan-martinek, 2021-01-04:
It would be a huge improvement in ironing quality. Also ironing a bit more beyond the perimeters would help.

k1-801, 2023-03-05:
I wrote a longer explanation on how this *could* work in #9451, with a minimal threshold for continuous ironing (closed it as the concept is essentially the same)

DoubleStrike, 2023-12-12:
Not seeing any additional comments in a very long time.  Any feedback from the devs on this possibility?

Skipping ironing on an area when another layer will be printed there is especially bad on the Prusa XL when we are dealing with mixed nozzle sizes.  Just having the ironing pass go across 100% of the surface and then printing the next toolhead's output onto a clean, ironed surface would yield **dramatically** better results.

RajHid, 2025-02-25:
I also would find it usful to have more control over the ironing.
I wanted to print a thin belt from [1] and the yigyag pattern disruptetd the ironing over the full length.
Ironing over the wohle belt should improve the adhesion between the layers and the layerlines and therfore improve the tensile strength of a part.
Also it would be an improvement for tensile strenth if there were an option to set a value different from 90° for the infill patterns.

When considering printing whith an carbon fiber material ironing in the direction of the expected load should align a certain amount of fibers in that direction.

[1] https://www.printables.com/model/122320-one-for-all-cable-strap-print-in-place-clamp-belt

k1-801, 2026-09-04:
Still relevant.