prusa3d/PrusaSlicer #5410: Ironing untick/tick bug

Frederic3d, 2020-12-08:
### Version
Prusa Slicer 2.3.0-beta2

### Operating system type + version
Windows 8.1, windows 10

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3 / 2018

### Behavior
Bug 1 :
- Enabling 'ironing'
- Setting the top pattern to other than 'Monotonic' doesn't warn that ironing is disabled

Bug 2 :
- Enabling 'ironing'
- Setting the top pattern to other than 'Monotonic' 
- Click on 'slice now'
- Go back to top pattern, put 'Monotonic' again
- Click on 'slice now'
=> The slicing doesn't do ironing => you have to untick 'Enable ironing', go back to preview, then tick again 'Enable ironing' to have it work again

5teveO, 2020-12-17:
This was a bug in the alpha build that was fixed.. looks like it has reappeared

lukasmatena, 2020-12-18:
Just for reference: the older issues similar to this one were:
- Ironing flag ignored on infill rotate #4937
- Ironing does not take effect when you change Top Fill Pattern. #4900
- Ironing layer not shown in preview when settings are changed #4510

lukasmatena, 2020-12-22:
Should be fixed in [2.3.0-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc2) by 8f3baeb. Please retest and let us know if it isn't.
Thanks for the report. Closing.

closed 2020-12-22 by lukasmatena