prusa3d/PrusaSlicer #5411: Ironing : angle of ironing

Frederic3d, 2020-12-08:
### Version
PrusaSlicer 2.3.0-beta2

### Operating system type + version
Windows 8.1

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3 / 2018

### Behavior
_new feature request_

Suggestion 1 : (or maybe I didn't find on my own, but I searched)
At the moment, I couldn't find a way to choose the angles for ironing. It is basically "vertical" or "horizontal" vs the plate. It can't be set for example to +/- 45° (diagonally)
Only way is to rotate the piece on the plate, which is not always possible
when you print something that's like 25cm long (you have to put it diagonally on the plate, so no way to rotate it otherway)
[image]

Suggestion 2 / or question, that is :
Would it be possible to set the pattern to something else than monotonic ? I suspect it would defeat the purpose, but I have the intuition that Hilbert Curves 

would achieve a better result for pieces that have one of the two X or Y dimension very smaller than the other. On very large surfaces, I'm less sure, but the hilbert curves are a little more 'chaotic', which also means that the small imperfections would get even less noticed ?

5teveO, 2020-12-17:
I believe its actually relative to the solid infill. If you have that at 0 or 90 degrees the ironing will then be at 45deg

Steve

edwardhorsford, 2022-09-16:
Found this in search of being able to set the ironing angle. I'm [following this guide for printing clear PETG](https://www.printables.com/model/15310-how-to-print-glass), which suggests having all infill layers going in the same direction. You can do this with aligned rectilinear infill, but the ironing is 45º offset from whatever the infill angle is.

My temporary solution is a modifier for the very last layer that changes the ironing angle - this works, but is somewhat arkward.

pavolNovak, 2023-10-26:
+1 on ironing angle which is not dependent on previous layer angle

closed 2026-09-29 by prusaslicer-bot[bot]