prusa3d/PrusaSlicer #5415: Preview with volumetric extrusion wrong

repetier, 2020-12-09:
### Version
2.3 Beta 2

### Operating system type + version

OsX Big Sur
### 3D printer brand / version + firmware version (if known)

Not relevant - Marlin

### Behavior

When objects are sliced with volumetric extrusion the shown extrusion width is wrong. It is neglecting the fact the e is volume and not distance in backward computation.

#### Project File (.3MF) where problem occurs

All

#### Possible Solution

I think the problem is in void GCodeProcessor::process_G1(const GCodeReader::GCodeLine& line) in this block:

  ```
  if (m_extrusion_role == erExternalPerimeter)
            // cross section: rectangle
            m_width = delta_pos[E] * static_cast<float>(M_PI * sqr(1.05 * filament_radius)) / (delta_xyz * m_height);
        else if (m_extrusion_role == erBridgeInfill || m_extrusion_role == erNone)
            // cross section: circle
            m_width = static_cast<float>(m_filament_diameters[m_extruder_id]) * std::sqrt(delta_pos[E] / delta_xyz);
        else
            // cross section: rectangle + 2 semicircles
            m_width = delta_pos[E] * static_cast<float>(M_PI * sqr(filament_radius)) / (delta_xyz * m_height) + static_cast<float>(1.0 - 0.25 * M_PI) * m_height;

```

You are using delta_pos[E]  but in case of volumetric extrusion you need to divide the value with pi*filament_radius^2 to correct for the included volumetric conversion. Or adjust equations.

Maybe there is another case when you import directly and not from gcode. You will know better.

bubnikv, 2020-12-11:
BTW I suppose you are setting M200
https://marlinfw.org/docs/gcode/M200.html
either in the custom start G-code header or in the firmware, right?

I wonder how the poor slicer could be aware of M200 if it is not contained in the G-code header.

repetier, 2020-12-11:
How slicer should know - well you have this check button in printer configuration to use volumetric e. That is how detected this. For preview if no M200 is present (is wasn't in my test) it should use the filament diameters configured. At least that is what I use in server now for gcode preview calculation. They are in the comments at the bottom so also known when reread from disk.

Actually it is not to expected to contain M200 in gcode. When you have configured printer for volumetric extrusion you do this to be independent from slicer diameter settings for later print so it matches the filament you have mounted and would define before starting print. Otherwise it would be the same as having regular extrusion and diameter defined in slicer so it only fits a special diameter.

I have attached a sample preview where it quite obvious since I change extrusion width mid print:
[xyzCalibration_cube volumetric.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/5679293/xyzCalibration_cube.volumetric.gcode.zip)

repetier, 2020-12-11:
BTW: Do you read the tail of gcode for preview/analysis first? In my preview generation I read first 1000 lines to detect which slicer it is and get infos contained in header, then I read the last 25kb of gcode to extract settings stored there. Only then I start interpreting gcode and hence I know it was volumetric and which diameters were used while slicing.

bubnikv, 2020-12-11:
> well you have this check button in printer configuration to use volumetric e. That is how detected this. For preview if no M200 is present (is wasn't in my test) it should use the filament diameters configured

Thanks.

> BTW: Do you read the tail of gcode for preview/analysis first? In my preview generation I read first 1000 lines to detect which slicer it is and get infos contained in header, then I read the last [...]

Thanks again. We reading the comments indeed.

Sorry, I am running for a 2nd month in a row in a "release" mode and am too tired to think clearly.

repetier, 2020-12-11:
No problem. I'm in same situation. Wanted to release server and continue finding issues:-) Also this is not critical just wanted to tell you since I noticed it while testing how I cope with new comments.

The new release is a big step I think. I always wanted it to produce detectable extrusion widths. Also I still prefer the simplify3d method most where the slicer adds:
; tool H0.200 W0.480
so for preview you get exact dimensions regardless of equation used and no need to know if equations have changed between versions and all other problems you have back computing the width from height and E which might inprecise for short moves with low layer heights. Simple and future proof.

lukasmatena, 2020-12-18:
This should be fixed (dcd69b4) in [2.3.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc1). Could you please retest and close the issue if happy? Thanks.

repetier, 2020-12-18:
My test gcode shows no more errors, so I also think it is resolved.

closed 2020-12-18 by repetier