prusa3d/PrusaSlicer #5416: High CPU usage when PrusaSlicer is in background

jwhevans, 2020-12-09:
### Version
PrusaSlicer Version 2.3.0beta2+

### Operating system type + version
macOS Big Sur
Version 11.0.1

### 3D printer brand / version + firmware version (if known)
Prusas MK3S latest firmware release 

### Behavior
After completing a slice and gcode export the program runs continuously at 12-15% CPU usage and 36% GPU usage

This happens for all projects used in the beta version. The release version does not maintain the high CPU/GPU usage.

I would expect the CPU usage to drop after the gcode is exported, particularly if the program is left in the background.

[image]
[tube v3.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5667633/tube.v3.3mf.zip)

rtyr, 2020-12-09:
Most likely a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5402.

jwhevans, 2020-12-09:
I believe you are correct. I have tested again on several files. If I leave the notification open the CPU core and GPU core run continuously. Closing the notification after export results in the CPU/GPU utilization dropping to 0

ojamin, 2020-12-17:
I left 2 instances running on the Print Settings tab with no notifications open and both instances minimised after an export. An hour later and they were each using 10% of my cpu unchanging. I then switched them to the Printer Settings tab and the cpu use dropped to 0%.

lukasmatena, 2020-12-18:
Could you please test in [2.3.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc1) and see if any of the issues with high CPU usage remain? Thanks.

ojamin, 2020-12-22:
> Could you please test in [2.3.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc1) and see if any of the issues with high CPU usage remain? Thanks.

I've tested various combinations on rc1 and can't replicate the issue now!

closed 2020-12-22 by lukasmatena

readysetawesome, 2021-04-25:
I just experienced this on the release version PrusaSlicer 2.3.0+

Computer was hot when I woke this morning, after being on all night with Prusa in the background, which was running 100% CPU.

[image]