prusa3d/PrusaSlicer #5418: Prusa slicer 2.3.0 BETA 2 crash when attempting to save a machine change

italocjs, 2020-12-09:
### Version
2.3.0 beta 2 (download a few min ago)

### Operating system type + version
Win10 2004

### 3D printer brand / version + firmware version (if known)
Unrelated 

### Behavior
I'm attempting to create a new printer (not using prusas) and when i click the save button i get the following error:
>>Unhandled unknown exception; terminating the application.
>>Internal error: boost::bad_format_string: format-string is ill-formed

This is happening with any printer i try to add, with default custom setting.

[image]

italocjs, 2020-12-09:
Just checked, same is happening in 2.3.0 beta1

zipped below is the .ini for the settings i tried to import.
[Configuracoes OFFICE - Atualizado em 31-01-2020.zip](https://github.com/prusa3d/PrusaSlicer/files/5668153/Configuracoes.OFFICE.-.Atualizado.em.31-01-2020.zip)

[image]

togomedia, 2020-12-10:
We had this problem and found it only happened when we tried to change the machine with objects on the build plate. My experience (Mac Catalina) is if you delete "all" build plate objects before changing the machine, the software does not crash. 

I know it's a workaround but may give insight into the issue.

bubnikv, 2020-12-10:
The issue was caused by typos introduced into Brazilian Portugese dictionary. Fixed with 
 https://github.com/prusa3d/PrusaSlicer/commit/63a7078432c6ece1deab66047133f96dc0a2147f

I have improved the error message with e7bf03f0064163112fef052a77265fbbb3ec2490, but we cannot really do much more than that, the application will have to terminate if that happens.
[image]

Closing.

closed 2020-12-10 by bubnikv