prusa3d/PrusaSlicer #5419: Y and Z inverted for the placement of an object

Lenny32, 2020-12-09:
### Version
2.3.0-beta2+win64 (obtained with the release zip folder for windows 64 bits)

The Y and Z are inverted while using the placement of an object 

(Blue is used as is showing Y and the arrow is wrong) 
[image]
(Green is the showing Z axis and the arrow is wrong)
[image]

You can the the highlighted blue box in the right corner of each pictures.

Lenny32, 2020-12-09:
After keeping on testing, I cannot reproduce the bug. Maybe something to keep in mind.

fe60, 2020-12-11:
I would close this issue and create a new one, once you are able to reproduce the bug.
Without an option to analyze and solve a concrete reproducible problem, this bug report will bloat up the backlog of unresolved issues.

Lenny32, 2020-12-14:
I am closing the issue.  I just wanted to report the bug.

closed 2020-12-14 by Lenny32