prusa3d/PrusaSlicer #5420: Fusion 360 opens new PrusaSlicer instance for each exported STL file

Ohmarinus, 2020-12-10:
### Version
PrusaSlicer 2.3.0-beta2

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Custom Printer/Prusa MK2.5/MK3 hybrid DIY clone with partial original Prusa parts.

### Behavior
When exporting STL files from Fusion 360 each time PrusaSlicer is loaded, a new instance is opened, this is highly impractical. I read somewhere a comment from a user on reddit that there is being worked in Fusion 360 integration:
https://www.reddit.com/r/prusa3d/comments/fqqzf6/fusion_360_to_3d_print_using_prusa_slicer_opens_a/fltz3lo/?utm_source=reddit&utm_medium=web2x&context=3
However, this is an old comment and nothing has changed yet.
Is there a way to get Fusion 360 to export into the same PrusaSlicer instance?

### Is this a new feature request?
Kind of.

Thanks for your wonderful work on the Slicer! I hope this issue/request finds it's way into a new edition.

bubnikv, 2020-12-10:
See https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-alpha1
Single vs. multiple instances, sending arguments to an existing instance

Ohmarinus, 2020-12-10:
> 
> 
> See https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-alpha1
> Single vs. multiple instances, sending arguments to an existing instance

Thanks, this works!!

I created a bat file with the following information in the PrusaSlicer directory, and am telling Fusion 360 to open the bat file to open an STL in PrusaSlicer:
`@start "" "%~dp0\prusa-slicer.exe" --single-instance %* &`
Filename: prusa-slicer.bat
The file needs to be put into the same directory as prusa-slicer.exe

Thanks for the addition! Cura also had this, but in Cura it doesn't work anymore.

closed 2020-12-10 by Ohmarinus

JayKeyStk, 2022-05-12:
@Ohmarinus I LOVE YOU! :-) :-P Been struggeling to find any solution to this!! Can't believe Prusa doesen't fix this behaviour... Thanks for the fix! :-)

bubnikv, 2022-05-12:
> Can't believe Prusa doesen't fix this behaviour... Thanks for the fix! :-)

How could we fix that? It is Fusion360 which does not allow one to specify command line parameters of the application to be started.

JayKeyStk, 2022-05-12:
By having PS behaving like it does on OS X?

There it always opens new files in the same window that’s already open ;-)

Best Regards
JayKey

puterboy, 2022-06-07:
> > See https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-alpha1
> > Single vs. multiple instances, sending arguments to an existing instance
> 
> Thanks, this works!!
> 
> I created a bat file with the following information in the PrusaSlicer directory, and am telling Fusion 360 to open the bat file to open an STL in PrusaSlicer: `@start "" "%~dp0\prusa-slicer.exe" --single-instance %* &`
> 
> Filename: prusa-slicer.bat The file needs to be put into the same directory as prusa-slicer.exe
> 
> Thanks for the addition! Cura also had this, but in Cura it doesn't work anymore.
> 
> **Update: to get this working in Fusion360, you need to manually type in the name of the `bat` file in the selection box in Fusion360 since Fusion only allows you to select `exe` files in choosing the 3D printing app!**

You can get around this by using one of the bat-to-exe converter programs available for free download and use (at least non-commercial).

bubnikv, 2022-10-11:
> Yeah? That would solve the issue wouldn't it? :-)
>
> You tell me :-) I'm not a programmer. I just know that it's several extra
> steps exporting meshes from 360 to PS on windows, which is missing from
> OSx. It's a lot smoother on OSx, more streamlined. And it bothers me :-P :-)

> You tell me :-) I'm not a programmer. I just know that it's several extra
steps exporting meshes from 360 to PS on windows, which is > missing from
OSx. It's a lot smoother on OSx, more streamlined. And it bothers me :-P :-)

The most sensible solution is to let Fusion360 implement PrusaSlicer
support. We are in touch with them, so a tighter integration with Fusion
360 may actually happen.

JayKeyStk, 2022-10-12:
NICE! 😁😀

Best Regards
Johan K Stokka

> You tell me :-) I'm not a programmer. I just know that it's several extra
steps exporting meshes from 360 to PS on windows, which is > missing from
OSx. It's a lot smoother on OSx, more streamlined. And it bothers me :-P :-)

The most sensible solution is to let Fusion360 implement PrusaSlicer
support. We are in touch with them, so a tighter integration with Fusion
360 may actually happen.

čt 12. 5. 2022 v 16:53 odesílatel JayKeyStk ***@***.***>
napsal: