prusa3d/PrusaSlicer #5422: encoding problem wit Russian Language?

userosos, 2020-12-10:
Hi! see pic.
[image]

bubnikv, 2020-12-10:
@repetier

We are receiving and parsing a JSON containing the groups here:
https://github.com/prusa3d/PrusaSlicer/blob/740ffc03b415be46edc7112ce1778d5f3346b0f7/src/slic3r/Utils/Repetier.cpp#L199

C++ is quite finicky in regard to string encoding, but the code looks good to me. We expect the group to be UTF8 encoded. We parse the JSON, extract the groups, convert from UTF8 to wxWidgets string, which is a 16 bit encoding on Windows. We then pass the 16 bit encoding to a wxWidgets list box.

How do you encode Unicode into the JSON provided for the "printer/api/port" query?

repetier, 2020-12-10:
Since we provide all data for browsers who also expect utf-8 we only use 8 bit strings everywhere and assume them to be utf-8 formated. For testing I added a group "Тестовая группа" and it appeared as that in your beta2 using BigSur and german language setting.

userosos, 2020-12-11:
I can see it in Settings Filament/dependens/select printers or profiles printing.
[image]

YuSanka, 2020-12-14:
> I can see it in Settings Filament/dependens/select printers or profiles printing.
> [image]

Duplicate of  #4975. Solved.

Kachidoki2807, 2020-12-18:
Hello,

I can also raise up an encoding issue with the "vendor profiles" (.ini under resources/profiles).
I have a printer nammed MakerFr i3 RS³² which is displayed like below in UTF8 under the Configuration Wizard:
[image]

In other hand, if I encode the .ini with Windows 1252 encoding, the text is properly displayed:
[image]
But the counterpart is that the corresponding thumbnail (also nammed with ³²) is not displayed anymore.

Tested under Win10.

lukasmatena, 2020-12-22:
@Kachidoki2807 The issue in ConfigWizard should be fixed in [2.3.0-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc2). Please retest and let us know if it isn't.

Kachidoki2807, 2020-12-23:
Hi @lukasmatena,
The ConfigWizard is fixed on my side . The text in the dialog is ok, and the thumbnail is displayed.

However, it is probably nothing, but I got something weird at the first trial. I re-generated the whole solution from the rc2 tag, and the text was ok but not the thumbnail. After to have manually removed the profile from AppData, done a new build and placed a breakpoint on the load_bitmap path evaluation, everything was ok this time. I have tried various way of cleaning and build, it never happened again. Probably I've done a mistake the first time. :-)

BTW (bit off-topic), is it normal that the bitmap are not transferred under AppData? The code always evaluate the AppData path to find the bitmap first, then go to find it into the resources directory, and finally display the no_thumbnail picture if so. The first line is always false as there are no bitmap stored into AppData.

https://github.com/prusa3d/PrusaSlicer/blob/d2b45c282e70ae75e1f99c158103833a667e0e5c/src/slic3r/GUI/ConfigWizard.cpp#L208-L209

bubnikv, 2021-01-04:
> BTW (bit off-topic), is it normal that the bitmap are not transferred under AppData? The code always evaluate the AppData path to find the bitmap first, then go to find it into the resources [...]

That is all right. The alternate "override" location is checked, but normally the pictures are not copied.

lukasmatena, 2021-01-04:
I hope that all problems mentioned here are now solved in [2.3.0-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc2). Closing the issue.
If anything remains, let us know. Thanks.

closed 2021-01-04 by lukasmatena