prusa3d/PrusaSlicer #5423: Prusa Slicer 2.2.0 assertion failures on startup when in HDR mode

chrisniggel-okta, 2020-12-10:
### Version
PrusaSlicer
Version:   2.2.0+win64
Build:     PrusaSlicer-2.2.0+win64-202003211145
### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 8,532MB
OpenGL installation
GL version:   4.6.0 NVIDIA 391.25
Vendor:       NVIDIA Corporation
Renderer:     GeForce GPU/PCIe/SSE2
GLSL version: 4.60 NVIDIA

NVIDIA System Information report created on: 12/10/2020 08:54:57

[Display]
Operating System:	Windows 10 Home, 64-bit
DirectX version:	12.0 
GPU processor:		GeForce GPU
Driver version:		391.25
Direct3D API version:	12
Direct3D feature level:	11_0
CUDA Cores:		640 
Core clock:		1019 MHz 
Memory data rate:	5010 MHz
Memory interface:	128-bit 
Memory bandwidth:	80.16 GB/s
Total available graphics memory:	6116 MB
Dedicated video memory:	2048 MB GDDR5
System video memory:	0 MB
Shared system memory:	4068 MB
Video BIOS version:	82.07.32.40.78
IRQ:			Not used
Bus:			PCI Express x16 Gen2
Device Id:		10DE 1392 066A1028
Part Number:		2704 0011

[Components]

NvUpdt.dll		10.4.0.5		NVIDIA Update
nvui.dll		8.17.13.9125		NVIDIA User Experience Driver Component
nvxdplcy.dll		8.17.13.9125		NVIDIA User Experience Driver Component
nvxdbat.dll		8.17.13.9125		NVIDIA User Experience Driver Component
nvxdapix.dll		8.17.13.9125		NVIDIA User Experience Driver Component
NVCPL.DLL		8.17.13.9125		NVIDIA User Experience Driver Component
nvCplUIR.dll		8.1.940.0		NVIDIA Control Panel
nvCplUI.exe		8.1.940.0		NVIDIA Control Panel
nvWSSR.dll		23.21.13.9125		NVIDIA Workstation Server
nvWSS.dll		23.21.13.9125		NVIDIA Workstation Server
nvViTvSR.dll		23.21.13.9125		NVIDIA Video Server
nvViTvS.dll		23.21.13.9125		NVIDIA Video Server
nvDispSR.dll		23.21.13.9125		NVIDIA Display Server
NVMCTRAY.DLL		23.21.13.9125		NVIDIA Media Center Library
nvDispS.dll		23.21.13.9125		NVIDIA Display Server
PhysX		09.15.0428		NVIDIA PhysX
NVCUDA.DLL		23.21.13.9125		NVIDIA CUDA 9.1.84 driver
nvGameSR.dll		23.21.13.9125		NVIDIA 3D Settings Server
nvGameS.dll		23.21.13.9125		NVIDIA 3D Settings Server

### 3D printer brand / version + firmware version (if known)
Prusa MK3S 3.9.2

### Behavior
* _Describe the problem_
PrusaSlicer shows assertion failures on startup when Windows is configured to HDR mode.  If "Continue" is clicked through all failures, Slicer starts, but no plate is displayed.

* _Steps needed to reproduce the problem_
Ensure HDR mode is turned off.
Start Slicer.  Slicer will start normally.
[image]
[image]
Right-click on desktop. Select "display settings"
Turn HDR Mode on
[image]
Start Slicer
Click "Continue" on each assertion failure.
[image]
[image]
[image]
Slicer will start with no plate shown.
[image]
Close Slicer
Right-click on desktop. Select "display settings"
Turn HDR Mode off
Start Slicer.  Slicer will start normally.
* _Expected Results_
Slicer should start normally
* _Actual Results_
Slicer has assertion failures and will not start.

_Is this a new feature request?_
No

bubnikv, 2021-01-06:
Do you get these issues with PrusaSlicer 2.3.0-rc2?
https://github.com/prusa3d/PrusaSlicer/releases

ChrisCrewdson, 2021-01-07:
PrusaSlicer 2.3.0-rc2 fixes this for me.

bubnikv, 2021-01-07:
thanks, closing

closed 2021-01-07 by bubnikv