prusa3d/PrusaSlicer #5424: Error Slicing STL File

dhladky, 2020-12-10:
### Version
_2.2.0+win64
Build: PrusaSlicer-2.2.0+win64-202003211145
### Operating system type + version
Windows 10 Home 64-bit (10.0, Build 18363) (18362.19h1_release.190318-1202)

### 3D printer brand / version + firmware version (if known)
MK3S 
Firmware 3.9.2 

### Behavior
* After fresh start import given STL
* The imported STL renders correctly (green characters)
* Slice the project with default settings for PLA (Prusament)
* The displayed slicing shows completely invalid result

[image]
[image]

#### Imported STL File
[abc 2cm plancha.zip](https://github.com/prusa3d/PrusaSlicer/files/5674978/abc.2cm.plancha.zip)

rtyr, 2020-12-10:
The walls are too thin to be printed with default extrusion width. Either enable the "Detect thin walls" option or reduce extrusion width of external perimeters to 0.4mm.

dhladky, 2020-12-11:
You are absolutely right! Thank you for very fast response. Closing the ticket.