prusa3d/PrusaSlicer #5426: Problem on the first layers

moreirae, 2020-12-10:
### Version
Prusa Slicer 2.3.0 Beta 2

### Operating system type + version
Microsoft Windows 10 Pro
10.0.18363 Compilação 18363

### 3D printer brand / version + firmware version (if known)
Creality Ender 3 Pro
Marlin 2.0.7.2

### Behavior
Problem on the external perimeters on the second layer, have a big diference from second and not stick and fall down.

[image]
[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[image]
[image]

rtyr, 2020-12-10:
Too big overhang on the model. Using more than 2 perimeters may help. I don't see anything unexpected here.

rtyr, 2021-02-26:
Anwered. No response. Closing.

closed 2021-02-26 by rtyr