prusa3d/PrusaSlicer #5428: how to eliminate top layer

rodokas, 2020-12-11:
### Version
2.2

### Operating system type + version
windows 10

### 3D printer brand / version + firmware version (if known)
COREXY own construction

### Behavior
In many prints that I use at the same speed in internal layers and in top layers, I realize that there is a very large waste of time depending on the piece, because the Prusa slicer creates the top layer separately, even if they have the same speed. Is there a way to eliminate the top layer, so that the slicer does not treat it as a separate item from the inner layers?

* _Steps needed to reproduce the problem_
in a slicing with very angles or thin walls, in many moments you will have an internal fill and a top fill in the same layer. when we use the same speed for both, there is no reason for the slicer to calculate and print as separate items, this greatly increases the printing time.

* _Expected Results_
a function to disable the construction of the top fill separately from the inner fill when they exist in the same layer.
* _Actual Results_
construction of internal filling and top filling separately in the same layer, even if the extrusion speeds and width are the same.
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
in the image below I would like the red and purple construction to be a single construction because in many cases I use the same speed.
[image]

_Is this a new feature request?_ YES

bubnikv, 2020-12-11:
If they have the same parameters, they should be merged into a single region. Often they differ with extrusion width.

rodokas, 2020-12-11:
I put all the parameters exactly the same, but they are still different regions

rodokas, 2020-12-11:
[image]
in this image you can see, all these regions are exactly the same settings

closed 2026-09-29 by prusaslicer-bot[bot]