prusa3d/PrusaSlicer #5429: Creating a printer profile named "Original Prusa i3 MK3S" crashes PrusaSlicer

spiff72, 2020-12-11:
### Version
_Version of PrusaSlicer used goes here_
2.3.0 Beta2+win64

### Operating system type + version
Windows 10
Version 20H2
Build 19042.685

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S MMU2S

### Behavior
* _Describe the problem_
I was cleaning up some of my printer profiles - removing some that were duplicates, etc., and I had one called "Original Prusa i3 MK3S - Copy".  I was trying to get rid of any unneeded "copy" names, so I saved that one as "Original Prusa i3 MK3S", with the intent of then going back and deleting the one with the - Copy suffix.  I thought this was safe since there was no other profile with this name.  Upon clicking the button to save it, the program froze and had to be killed in the task manager.  I tried to restart it, and it hung on the splash screen.  I presumed that this may be a reserved name since it was a default name previously (now replaced with the extra "+" names in the profiles).  I then went into the AppData/Roaming/PrusaSlicer-beta/printer folder and found the "Original Prusa i3 MK3S.ini" file that had been saved there a few minutes ago, deleted it, restarted PrusaSlicer and it started back up normally.
* _Steps needed to reproduce the problem_
See above for steps to reproduce - save a printer profile with the name "Original Prusa i3 MK3S".
* _Expected Results_
That profile name should either be flagged as a reserved name and protect a user from their own stupidity, or it should be capable of being named such.

_Is this a new feature request?_
No

spiff72, 2020-12-11:
I should add that this is probably not a critical bug - as I stumbled across it by accident.  But if anyone like me has some older print profiles that originated from a time before they had the MMU2S upgrade, this might crop up on rare occasions.

Just for grins, I tried doing it again, but left a space at the end of the profile name, and no crash.  But every time I tried to duplicate the issue it would occur every time.

FidelCapo, 2024-07-15:
This issue should be already fixed in the released version 2.8.0.
Closing.

closed 2024-07-15 by FidelCapo