prusa3d/PrusaSlicer #5430: Crash of SW 2.3.0-beta2+64 with RAFT-Setting

kingfisherXL, 2020-12-11:
### Version
Prusa 2.3.0-beta2+win64
Language: German

### Operating system type + version
Win10 Professional

### 3D printer brand / version + firmware version (if known)
Anycubic I3 Mega S; Marlin 1.1.9.1

### Behavior
* RAFT setting is 2 layers
* the stl-files is a simple net / grid structure (see screenshot)
* Slicing a a ends up in a crash of the SW.
   (start slicing the -> A Warning is displayed - like "empty layers detected" -> a dialog box appeares "Pruca crashed, debugging..")
* Slicing the same stl-file with prusa 2.2.0 is showing the same error message, but does not end up in a Crash !

The behaviour is reproduceable !! 
The main issue - is the RAFT setting;
Deleting it - does end up in a normal slicing

-> BUG
[image]

rtyr, 2020-12-11:
Please provide your 3MF project file.

kingfisherXL, 2020-12-11:
sorry - forgotten - here it is the 3mf file (within the zip)
[CalSqrNet_Any.zip](https://github.com/prusa3d/PrusaSlicer/files/5681090/CalSqrNet_Any.zip)

bubnikv, 2020-12-11:
I am able to reproduce the crash. Thanks for the report.

bubnikv, 2021-08-31:
This issue seems to be fixed in PrusaSlicer 2.4.0-alpha1. Closing.

closed 2021-08-31 by bubnikv