prusa3d/PrusaSlicer #5432: Exporting chitubox format?

eried, 2020-12-11:
### Version
2.2.0

### Operating system type + version
win64

### 3D printer brand / version + firmware version (if known)
creality ld-002r

### Feature request
Are there plans for supporting another types of export? chitubox format for example?

bttguy, 2021-01-14:
+1 if possible, a lot of us would be thankful for that :)

lukasmatena, 2022-03-21:
Also #5280. In fact, I think this one should be closed in favor of that one? Right?

lukasmatena, 2022-04-05:
Apparently.

closed 2022-04-05 by lukasmatena

NickSica, 2023-04-25:
In late February I picked this up and made some distance with it, but stopped due to changes in the codebase. It seems that was a good idea as the pwmx I was using as reference was changed quite a bit. @Mimoja is there any insight on any big changes(mostly to implementing the AnycubicSLAArchive as that was one of the new features added). Also I added support through the GUI instead of app notes in my changes, is there a reason this isn't done for the Anycubic printers? Should I go through app notes instead?