prusa3d/PrusaSlicer #5433: [Feature request] Reduce speed on high command rate

DBa2016, 2020-12-11:
### Version
Prusaslicer 2.2.0

### Operating system type + version
Win 10
### 3D printer brand / version + firmware version (if known)
Tevo Tornado running Klipper 0.9.0
### Behavior
When slicing https://www.thingiverse.com/thing:2795194 (for a 1.0mm nozzle) in vase mode, the amount of movements (G1 commands) in parts which are smaller than 50mm radius was exceeding my printers (or my Octopi's?) capability to process them. One could clearly see Klipper buffer overruns in the performance graph. This results in blobs on the print - reducing the speed makes the print smooth again.
While producing G2/G3 arcs would be the most elegant solution, the support for vase mode seems to vary across the firmware versions (due to the non-horizontal alignment of the arcs). Another solution would be to reduce printing speed to make sure the command rate per second does not exceed the threshold. Ideally, this should happen only in the parts which are small enough to produce this stuttering (to not increase the overall printing time excessively).

This is a new feature request.

sslupsky, 2021-02-17:
See also https://github.com/prusa3d/Prusa-Firmware/pull/2657.  Perhaps the parent of this PR (Octoprint Arcwelder plugin) could be integrated into slicer?

sslupsky, 2026-09-02:
I would have expected a major version of PrusaSlicer to address the fundamental CAD modelling and move away from using mesh as the underlying modelling.  Rather, the modelling should done using a reliable open source model such as BREP.  Where someone generates a model and cannot produce a BREP/STEP file, then interpolation of a mesh would be acceptable.

Thus, this issue still needs to remain open since G2/G3 generation is still not very good in 2.9.6.  It helps a bit but generally I see a huge number of linear movements in the gcode where they could be reduced to one arc.

Jan-Soustruznik, 2026-10-06:
Hello, thank you for the update. Following our new policy, I am closing this feature request and converting it into a Discussion thread. This will allow us to hear what the community thinks about this feature and provide a place for others to upvote it. Have a great day.