prusa3d/PrusaSlicer #5434: Adaptive Support Cubic mentioned in release notes, but not found in latest PrusaSlicer 2.3.0-beta2

mvadu, 2020-12-11:
### Version
PrusaSlicer 2.3.0-beta2

### Operating system type + version
Windows 10

### Behavior
PrusaSlicer 2.3.0-alpha1 release notes mention "We have added an "Adaptive Support Cubic infill" variant," but in the latest beta infill drop down only has Adaptive Cubic or Support Cubic, not an Adaptive Support Cubic. Am I misreading the release notes?

_Is this a new feature request?_
No

[image]

hejllukas, 2020-12-11:
Hi, you don't misread the release note. We use shortened the name in the slicer, but we left the full name in the release node. So Support Cubic is Adaptive Support Cubic. I admit that it can be a little bit confusing.

mvadu, 2020-12-11:
@hejllukas Thank you for the clarification. I will go ahead and close this issue.

closed 2020-12-11 by mvadu