prusa3d/PrusaSlicer #5436: Feature types selection no longer available on Feature type view

limited660, 2020-12-11:
### Version
_2.3.0-beta1+win64_

### Operating system type + version
_Windows 10 Pro - 64 bit - 20H2_

### Behavior
_Toggling of feature types is no longer a selection on the "Feature Type" view where it's most useful as they are color coded. This existed on previous versions as seen in the screenshot on [issue 3112](https://github.com/prusa3d/PrusaSlicer/issues/3112)._

_Current View_
[image]

_View in other types and previous versions of PrusaSlicer_
[image]

### Is this a new feature request?
_No_

bubnikv, 2020-12-12:
You can control visibility of features by clicking at the legend.

rtyr, 2021-02-26:
Is it understandable? Can we close the issue?

rtyr, 2021-04-12:
Answered, no response. Closing.

closed 2021-04-12 by rtyr