prusa3d/PrusaSlicer #5437: Always show the + layer icon

tonypartridge, 2020-12-12:
### Version
>= 2.2 

I’ve noticed an issue for me and a few over users over Facebook. We don’t see the + icon on the layer view. This is removed when print individual objects is enabled 

Instead we should show the + icon but stating you cannot set layers when individual object printing is enabled.

fe60, 2020-12-12:
Greying out the `+` next to the layer slider of the preview would indeed make it more obvious to users, that there is a feature, but it can't be used. A tool tip telling, that it's disabled because of the sequential printing, would make it perfect.
In addition either the documentation for Sequential Printing [1] or the one about Color Change [2] should mention, that one can't be combined with the other.

[1] https://help.prusa3d.com/en/article/sequential-printing_124589
[2] https://help.prusa3d.com/en/article/color-change_1687

piotr-kubiak, 2021-01-17:
I spend two hours trying to figure out why the (+) button is missing, only to realize that it is not compatible with Complete individual objects. It would be great if the button was greyed out instead of missing.

rtyr, 2021-08-11:
> Greying out the + next to the layer slider of the preview would indeed make it more obvious to users, that there is a feature, but it can't be used. A tool tip telling, that it's disabled because of the sequential printing, would make it perfect.

Implemented on master, it will be included in the upcoming PrusaSlicer 2.4.0-alpha1. Closing.

closed 2021-08-11 by rtyr