prusa3d/PrusaSlicer #5439: Extruder stops turning after printing solid fill around perimeters (picture below)

Zommy5re, 2020-12-12:
### Version
_2.3.0 beta1_

### Operating system type + version
windows 10 64bit
### 3D printer brand / version + firmware version (if known)
Anycubic i3 mega-S, lots of mods, modified firmware, but it prints these fine on 2.2.0
### Behavior
*Once print started, goes perfectly fine until it prints parts like picture attatched, which after only a few moves completely stops my extruder (e3d titan) from even trying to extrude, it never comes back on during the rest of the print. 
* print this: https://www.prusaprinters.org/prints/27216-bag-clip-print-in-place-cam   with this config: 
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_no_

#### Project File (.3MF) where problem occurs
_
[i3mega-S extruder issues.zip](https://github.com/prusa3d/PrusaSlicer/files/5682868/i3mega-S.extruder.issues.zip)

_
[image]

rtyr, 2020-12-13:
Likely hardware/firmware issue.

rtyr, 2020-12-15:
I printed the gcode from your 3MF project without issues, so your issue is likely not PrusaSlicer related. Closing.

[image]

closed 2020-12-15 by rtyr