prusa3d/PrusaSlicer #5441: 2-colour print stopped after done with priming only?

deonholt, 2020-12-12:
### Version
_2.3.0 beta2_

_2.3.0-bate2+win64_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_Win 10 64-bit_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_Creality Ender 5 Pro, Duet2 WiFi, 2 extruders, Y-split Creality Hotend_

### Behavior
* _Print stops after priming_
* _I send the g-code from PS to Duet via Wifi; it does G28 G29 and prints prime lines well and stops print after this. It shuts down hotend and bed as if print is done. When this happens, the hotend stops as on picture. After the 1st time this happened, I sliced agian and had the exact same thing happening._
  * _not sure what causes this_
* _complete print_
* _prints stops after priming the hotend_
  * _Screenshots from __*PrusaSlicer*__ p
[image]
[image]
[dual_extrusion_calibration_cube.zip](https://github.com/prusa3d/PrusaSlicer/files/5683015/dual_extrusion_calibration_cube.zip)

review are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2020-12-13:
It is likely caused by "M1 S10" command at the end of the priming. The "Prime all printing extruders" option was added specificaly for our multimaterial upgrade (Marlin based firmware). M1 command works differently on RepRapFirmware.

You have to disable initial priming for now.

[image]

deonholt, 2020-12-13:
Thanks. Will do it for now. I believe beta3 is out. Will download it and check again. 👍

closed 2020-12-16 by bubnikv

lukasmatena, 2020-12-16:
This is hopefully fixed with a5d363a. Could you please test in the next version after it is released (the one after beta3)? Thanks.

commit d85933b9 (2020-12-14): Do not emit M1 gcode for firmwares other than Marlin (after MM priming)
The M1 gcode is apparently only supported on Marlin, others do not support it or use it for something else
This should fix #5441
commit 788fbbc0 (2020-12-14): Do not emit M1 gcode for firmwares other than Marlin (after MM priming)
The M1 gcode is apparently only supported on Marlin, others do not support it or use it for something else
This should fix #5441
commit a5d363a9 (2020-12-14): Do not emit M1 gcode for firmwares other than Marlin (after MM priming)
The M1 gcode is apparently only supported on Marlin, others do not support it or use it for something else
This should fix #5441