prusa3d/PrusaSlicer #5443: Reload from disk unavaible for AMF

VorpalBlade, 2020-12-12:
### Version
2.3.0-beta2+linux+x64+GTK3

### Operating system type + version
Ubuntu 18.04

### 3D printer brand / version + firmware version (if known)
Irrelevant for bug (but Prusa MK3S).

### Behavior
* Import an AMF file exported from CAD software that doesn't support 3MF, such as FreeCAD.
* Update said file from the CAD software by re-exporting it after making some changes.
* "Reload from disk" is greyed out in the context menu of the object in PrusaSlicer, so the object has to be deleted and reimported manually, which might result in loosing settings.

_Is this a new feature request?_
No, this is a bug.

#### Project File (.3MF) where problem occurs
Happens without saving the 3MF as well as after saving it.

bubnikv, 2020-12-14:
Please attach the two versions of AMF, so we can retest. Thank you.

Raniz85, 2020-12-14:
I've got the same issue here. For STL files _Reload from disk_ is available, but not for AMF.
```
$ uname -a
Linux raniz-xps-15 5.9.13-arch1-1 #1 SMP PREEMPT Tue, 08 Dec 2020 12:09:55 +0000 x86_64 GNU/Linux
$ pacman -Qs prusa-slicer
local/prusa-slicer 2.2.0-9
    G-code generator for 3D printers (Prusa fork of Slic3r)
```

[hearts.zip](https://github.com/prusa3d/PrusaSlicer/files/5691028/hearts.zip)

hejllukas, 2020-12-15:
It should be fixed with https://github.com/prusa3d/PrusaSlicer/commit/d59f345fbb0e77e2fc6618766881c4ecb364c552

VorpalBlade, 2020-12-16:
> Please attach the two versions of AMF, so we can retest.

Sorry about that, but as the bug was apparently fixed already, do you still need this? I can of course provide one if it is still needed.

hejllukas, 2020-12-17:
If you can provide AMF file to us, we can test it if your issue has also been resolved. But I think your problem is the same as Raniz85's, which we fixed.

Raniz85, 2020-12-17:
Can confirm that the issue has been resolved in master

VorpalBlade, 2020-12-17:
I suepect the issue has indeed been solved, but here is a test AMF file: [test.tar.gz](https://github.com/prusa3d/PrusaSlicer/files/5710282/test.tar.gz)

lukasmatena, 2020-12-18:
This really seems to be fixed in [2.3.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc1). Let us know if it isn't. Closing the issue.

closed 2020-12-18 by lukasmatena