prusa3d/PrusaSlicer #5444: Dark theme broken with GTK3

VorpalBlade, 2020-12-12:
### Version
Version 2.3.0-beta2+linux+x64-GTK3

### Operating system type + version
Ubuntu 18.04 with the Cinnamon desktop environment.

### 3D printer brand / version + firmware version (if known)
Not relevant for this.

### Behavior
* Use a dark system theme in the GTK3 build of PrusaSlicer.
* Some text is unreadable due to being dark on dark.

The dialog asking for saving changes when closing a file or switching to a different profile:
[image]

The borders around controls that serve to group them are black on dark grey:
[image]

This control is broken, probably due to being too small. It is probably also broken with a light theme. Iit looks even worse if you select a longer type like NYLON.
[image]

The bed shape dialog is broken in two ways: Some input fields appear too short, and secondly the preview doesn't look like it has been adapted based on the system theme:
[image]

_Is this a new feature request?_
No, it is a bug.

#### Project File (.3MF) where problem occurs

fiechr, 2020-12-12:
How is the theme called exactly?

I've tested "Adwaita-dark" and "Breath-dark" on Manjaro Linux and it looks good.

VorpalBlade, 2020-12-12:
I had to look at this myself, been so long since I set it up. Apparently the theme is a mix of things in the Cinnamon settings:
[image]
[image]

I have used other desktop environments previously on this computer, so it might be possible there is a mix of settings being picked up? Though I haven't seen issues in any other GTK3 program (and GTK2 programs such as the other build of PrusaSlicer appears to use a light theme from somewhere).

I switched to Adwaita-dark, and restarted PrusaSlicer, but I see the exact same behaviour. Maybe that is the result of "Adwaita" + "Use a dark theme variant when available"? You likely have a newer version of all the system libraries though, as I'm stuck on Ubuntu 18.04 for now until certain projects at university migrate to a newer version of ROS.

VorpalBlade, 2020-12-12:
Just checked with beta3 that was just released an hour ago. Issues persist.

These issues are also in the GTK2 version if I use "Adwaita-dark" instead of "Adwaita" + "Use a dark theme variant when available".

In addition I see this black on dark text in the GTK2 build:
[image]
This was fixed for GTK3 in issue #4911, but apparently exists for GTK2 as well, I just never noticed it since I only ended up using the dark theme in GTK3.

fiechr, 2020-12-13:
I found that different GTK themes produce different results, although it shouldn't -- but I guess that's only in theory. The default Adwaita seems to work best. If I select other themes, the font in the drop-down menus is shifted and stuff like this. That maybe an issue of the themes, GTK2/3 itself, the framework PrusaSlicer is using (WxWidgets AFAIK) or PrusaSlicer itself.

With Adwaita-dark and GTK3 it looks like this for me:
[image]
[image]

Since I use KDE/Plasma I don't see any options to mix different styles and themes for GTK, I can only select a theme as a whole.

bubnikv, 2020-12-15:
@YuSanka fixed the dark mode issues.

Unfortunately we are scared to touch anything now before release for
[image]

because of 
https://www.youtube.com/watch?v=VoP1E9J4jpg

We would like to ask the Linux community to look into it, test thoroughly and provide a pull request.

YuSanka, 2020-12-17:
Thanks for your report. Most of those bugs were fixed and will be a part of upcoming release.

closed 2020-12-17 by YuSanka

VorpalBlade, 2020-12-17:
> Thanks for your report. Most of those bugs were fixed and will be a part of upcoming release.

What about the cut off text for filament type? My understanding is that this was not fixed. If so, was a separate bug opened for this?

YuSanka, 2020-12-17:
> What about the cut off text for filament type? My understanding is that this was not fixed. If so, was a separate bug opened for this?

Now, I'll fix it.