prusa3d/PrusaSlicer #5445: Russian Language - Please do not merge different version translations

AndylgTom, 2020-12-12:
**### Version**
PrusaSlicer-2.3.0-beta2+win64-202012051115

Please replace the current Russian translation with the one posted here https://github.com/prusa3d/PrusaSlicer/commit/19921e272a3b9d79aa1367a47d156bdb9f3d0842. 
**Or I also constantly upload updates to the cloud here https://yadi.sk/d/znelCpr5rgFV1Q . Last updated on 9.12.20.**
**And in any case, do not merge** it with the old translation which we made with @alexws54tk, otherwise it turns out nonsense and there are no connections in the translation of terms. Just open *. po and save it to *. mo (or get mine from the *.mo cloud).  Translated there exactly more than 95% and even what was not in the*. pot file.

**Also attached files (po, mo) here too from 9/12/20.**
[PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/5683536/PrusaSlicer.zip)

YuSanka, 2020-12-17:
OK, I would like to update your dictionary in master. But your work doesn't correspond to the last committed version of RU dictionary from [my last commit](https://github.com/prusa3d/PrusaSlicer/commit/3e0ce8a05420f2a6eff8a1ec54e388569cebc5fc).
So, all my fixes will be lost in that case.
And, please, look to the notes from @Goury [here](https://github.com/prusa3d/PrusaSlicer/pull/4390#issuecomment-739271366).
I'm agree with them. For example "copyright" is not "авторское право" but "право на копирование".

AndylgTom, 2020-12-17:
> И, пожалуйста, посмотрите заметки от @Goury [здесь](https://github.com/prusa3d/PrusaSlicer/pull/4390#issuecomment-739271366) .

I wrote the answer in that subject.

closed 2024-02-18 by lukasmatena