prusa3d/PrusaSlicer #5446: Auto-cooling should be based on time to next layer in object

jschuh, 2020-12-12:
### Version
2.3

### Behavior
Auto-cooling is presently based on the time to print the current layer. This means that when printing multiple objects with varying layer heights the auto-cooling flow will be unnecessarily applied to the object with larger layers. 

#### Actual Results
This issue is demonstrated in the screenshot below, which includes a cylinder on the left at 0.28mm layer height and a cube on the right at 0.05mm layer height. The red layers are where auto-cooling flow is used, and the blue layers are where normal flow is used because the layers between the two objects are aligned and the time for both objects gets counted in the layer time calculation.

#### Expected Results
I believe the correct behavior here would be to apply auto-flow not based only on the time for the current layer, but instead to segment the auto-cooling flow by object (when the layer heights differ), and apply it to each object based on the print time for the object plus the time until the next layer in that object will being printing. This would result in the cylinder in the image using only the normal flow and appearing entirely blue, while the flows for the cube would be the same as they are in the current implementation (due ti the smaller layer heights).

[image]

#### Project File (.3MF) where problem occurs
[auto_cooling_example.zip](https://github.com/prusa3d/PrusaSlicer/files/5681751/auto_cooling_example.zip)

bubnikv, 2021-01-06:
Also similar to https://github.com/prusa3d/PrusaSlicer/issues/984

jschuh, 2021-01-09:
Ah, I tried searching for an issue before filing this, but hadn't found that one. This is clearly a superset of #984, so I'm fine with merging this into that one.

bubnikv, 2021-02-16:
I have fixed #984, but this one is much more complex to be done right, so it will take much more effort, namely it is an optimization problem: Should the layers A be interleaved with layers of object B or vice versa? This is then much more complex for many objects and variable layer height.

closed 2026-09-29 by prusaslicer-bot[bot]