prusa3d/PrusaSlicer #5447: Custom colors won't save

nirvdrum, 2020-12-12:
### Version
Version 2.3.0-beta3+win64

### Operating system type + version

Windows 10 Pro

### 3D printer brand / version + firmware version (if known)

Prusa i3 MK3S + MMU2S
FW: 3.9.1

### Behavior

When I add a custom color by providing either HSL or RGB value, the color is not saved. If I re-open the color selection for a filament, the custom colors are always reset. Likewise, the "Define custom colors" button is always disabled.

_Is this a new feature request?_

No.

#### Project File (.3MF) where problem occurs

This happens with stock settings and with all projects.

nirvdrum, 2020-12-24:
I dug into this one a bit. I've never used wxWidgets before, so my analysis may be off. I think the core problem is the `wxColourData` values are never saved anywhere. Instead the two places where a `wxColourDialog` is created always create a new `wxColourData` instance, initializing it with whatever the extruder color should be. I think the reason for that is the extruder color is saved in the project file, but custom colors currently are not.

I should be able to submit a patch that would retain the custom colors for a live session. This would solve my immediate problem of trying to swap colors around between extruders with the MMU2S. But, those custom colors would not persist across sessions. That would require storing additional data in the printer config. I think that probably makes sense, since a user is apt to reuse colors, but I don't know if adding more metadata there is welcomed.

bubnikv, 2021-01-05:
The color picked at the filament combo box at Plater stores under extruder
color of the active printer profile.

closed 2026-09-29 by prusaslicer-bot[bot]