prusa3d/PrusaSlicer #5451: Scrollbar in plater view obstructs/overlaps content

Impact123, 2020-12-13:
### Version
`2.3.0-beta3+win64`

### Operating system type + version
`Windows 10 Pro`, `10.0.19042 Build 19042`

### Behavior
As can be seen in the screenshots, `2.3.0-beta3` not only creates a scroll bar even though there's enough space, but also obstructs the setting icons/buttons with it. This didn't happen with `2.3.0-beta2`.
* _Steps needed to reproduce the problem_
  * Enable `Expert mode`
  * Load a model
  * Slice it
  * Switch to plater view
  * Select/deselect the model
* _Expected Results_
Behavior as before where there is no scrollbar unless required and if there is one it should push the content instead of overlapping it
* _Actual Results_
  * `2.3.0-beta2`
    [image]
  * `2.3.0-beta3`
    [image]

Area5142, 2020-12-13:
I see the same problem on Linux 19.3 x64 but worse as the buttons is completely hidden when the scrollbar is on screen.

Just after start of PS:
[image]

When scrollbar is displayed:
[image]

I was sure there had been some design changes and the functionality was moved, but couldn't find it...

YuSanka, 2020-12-15:
Thanks for your report. Fixed with https://github.com/prusa3d/PrusaSlicer/commit/8627c35057373f14b0beb9f9b9b064a1141d2f09

closed 2020-12-15 by YuSanka

Impact123, 2020-12-18:
Thanks, but I'm afraid this isn't fully fixed and I'd consider this to be a regression in comparison to `2.3.0-beta2`.
The buttons/icons are now no longer obscured in `2.3.0-rc1`, which is nice, the scrollbar however is still unnecessarily shown even if there's unoccupied space. I reckon being able to see the **full** slice info is quite important and shouldn't be hidden on a 24" 1080x1920 (as in, not small or low resolution) screen with just a single object. I could scroll down, but the info will shift upon re-slice as explained in #5366.
Just for completion here's how my sidebar looks right now
[image]

lukasmatena, 2020-12-18:
This should be fixed in [2.3.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc1). Let us know if it isn't. Thanks.

Impact123, 2020-12-18:
I just did. The screenshot was made in `2.3.0-rc1`

Area5142, 2020-12-18:
@Impact123 see related issue #5462.