prusa3d/PrusaSlicer #5457: EINSTELLUNG MMUS2 MEHRFARBDRUCK WISCHTURM bei 0,25er Düse

Markushamm27, 2020-12-13:
### 2.2.0
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
Ich möchte ein Mehrfarbiges Schild ausdrucken. Düsendurchmesser 0,25mm. Bei der ersten Schicht benötige ich einen Durchfluss >70 um ein vernünftiges Ergebnis zu erzielen. wen dann ein Filamentwechsel ansteht und dazu der wischturm verwendet wird habe ich das muss ich die Durchflussgeschwindigkeit erhöhen, damit die Düse kpl. Gerienigt wird und nur noch das nächste Filament austritt.

Wer kann mir da helfen.

Kann ich über den G-CODE befehle einfügen, die dafür sorgen, das wenn der Drucker sich im Bereich des Wischturms befindet, der durchfluss erhöht wird. Wenn er dann wieder ausserhalb ist der durchfluss wieder reduziert wird.

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

fiechr, 2020-12-13:
Hier geht es allein um die Entwicklung der PrusaSlicer Software.

Die richtige Stelle für solche Fragen ist z.B. hier: https://forum.prusaprinters.org/
Oder auch hier: https://drucktipps3d.de/forum/forum/prusa-mk/

Markushamm27, 2020-12-14:
Hallo Christian,

vielen Dank die erste verständliche Antwort, ich versteh ja das ihr das
momentan aufgrund anderer Entwicklungen
nicht am Schirm habt.

Was für mich aber nicht verstündlich ist, daß ihr die PARAMETER für den
MMU2S single prozess habt. Wäre es nicht einfaches diesen
Daten auf MMU2S Multifarbenmodus zu übertragen.

Na ja trotzdem Danke für deine Antwort, ich werde selber mal probieren die
enstellungen so zu verändern, das es funktioniert.

l.G.

Markus

closed 2026-09-29 by prusaslicer-bot[bot]