prusa3d/PrusaSlicer #5460: PRUSA SLICER MMU2S MEHRFARBDRUCK EINSTELLUNGSPROFIL FÜR 0,25 DÜSE

Markushamm27, 2020-12-13:
### Version
_Version 2,2,0 WINDEOWS 64

out->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
WINDOWS 10

### 3D printer brand / version + firmware version (if known)
MK3S FIRMWARE 3.9.1-3519
MMU2S Firmware 1.0.6-372

### Behavior
* Ich möchte ein mehrfarbiges Objekt mit einer 0,25er Düse ausdrucken_
* _Steps needed to reproduce the problem_
Wie kann man kurzfristig so ein Profil erzeugen
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
Kann dieses Profil im nächsten SLICER UPGRADE includiert werden
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2020-12-13:
Already answered here https://github.com/prusa3d/PrusaSlicer/issues/3048. We are not planning 0.25mm nozzle MMU profiles in near future, since it would be not compatible with current ramming settings. Closing.

closed 2020-12-13 by rtyr

Markushamm27, 2020-12-13:
Why do you have it included in the mmu2s single mode, therfore im thinking
that is also for mmu2s Standard mode what is here the problem.

Markushamm27, 2020-12-13:
Ok thankyou but why do you have it in single mode

rtyr, 2020-12-13:
Because there are no toolchanges in the single mode.