prusa3d/PrusaSlicer #5461: Wipe feature inconsistent on small features

josephflat, 2020-12-13:
### Version
Version 2.3.0-beta3+win64

### Operating system type + version

Windows 10 +64bit
### 3D printer brand / version + firmware version (if known)

Creality Ender 3 + Marlin 2.0.x
### Behavior

On small features like a spire-shaped or column feature and there is a gap fill in the middle of it, the "Retract Amount Before Wipe" setting will not matter as it will try to wipe on that small gap fill area which is essentially not a wipe movement. this produces stringing or whisking on leaky extruders or stringy filaments which the setting is useful for. 
Then I went back to the Beta 2 and Beta 1 versions seeing them a little bit more consistent but sometimes are absent. 
Going back a little further in PrusaSlicer 2.2.0 without the tool position preview, but you can see the wipe movement as red lines being inconsistent as well. 

This is my first time posting an issue on github and i'm not sure if this has been tackled before but i've been trying to dial in questionable-quality PLA and PETG lately and upon closer look i think this is the only thing preventing me from getting good results and i've gone through all the ways to dial in retraction settings and I have had some success but when i try out new filaments, my wits get tested again. 

* _Steps needed to reproduce the problem_
Slice these retraction tests i've uploaded and notice the wipe lengths. Sometimes they are even absent even without any gap fill. 

  * _Screenshots, Sorry for the ugly markings_
[image]
[image]

#### Files - STL's, config, and 3mf
[Wipe Issue Files.zip](https://github.com/prusa3d/PrusaSlicer/files/5684835/Wipe.Issue.Files.zip)

espr14, 2022-06-16:
Looks like it does not appear on Prusa's original printers but on others. Sometimes the wipe is OK, sometimes it's few microns long and sometimes it's missing completely (no comment in G code).

Sverto, 2023-03-14:
Experiencing the same issue.
On the left there is no wipe, on the right there is:
[image]
Both sides have an equal size. Seems like a bug?

closed 2026-09-29 by prusaslicer-bot[bot]