prusa3d/PrusaSlicer #5462: Expert mode right panel avoidable scrollbar UI regression

pmjdebruijn, 2020-12-13:
### Version
PrusaSlicer 2.3~beta3

### Operating system type + version
Xubuntu 20.04 (tried with both stock wxGTK 3.0.4 and wxGTK 3.1.4)

### Behavior
After having sliced a model, with the Plater tab view in Expert mode (the other modes don't expose this issue), the right panel suddenly gets a scrollbar, eventhough there isn't a shortage of space, as the Object Listbox is large and empty.

I'm not perfect sure exactly when this regression was introduced, but it likely happened recent-ish.

Area5142, 2020-12-14:
I think it has to do with the issue I reported: _Object tree is too small for MMU/object manipulation_ #5162.
In 2.3.0 beta3 the object tree was changed to have a minimum size, not an actual and max size as expected. This results in a panel with wasted space when there are few objects in the list.

PhilMegman, 2020-12-15:
> ### Version
> PrusaSlicer 2.3~beta3
>
> ### Operating system type + version
> Xubuntu 20.04 (tried with both stock wxGTK 3.0.4 and wxGTK 3.1.4)
>
> ### Behavior
> After having sliced a model, with the Plater [...]

Comment to operating system:
I have the same issue on Windows 10.

Comment to Introduction:
I changed from alpha 2 to beta3, and alpha2 did not have this issue.

pmjdebruijn, 2020-12-17:
This regression was introduced in:
https://github.com/prusa3d/PrusaSlicer/commit/d7412d4e54c9339300ac8ff5175c70278fd0d55e
Reverting that commits restores the old behavior.

Area5142, 2020-12-18:
If d7412d4e54c9339300ac8ff5175c70278fd0d55e is reverted then issue **#5162 should also be reopened**.

When using the object list in MMU mode, the object list window is way too small. In MMU there is far more objects as there is an object for every color and assigning colors is done in the object list.

I suggest that the list resizes itself to the current object tree size but has a maximum size when the tree is large. That way the list tree only takes the size needed when there is few objects/entries but grows big enough to be usable when using MMU and/or many modifiers.

pmjdebruijn, 2020-12-18:
Confirmed Fixed

closed 2020-12-18 by pmjdebruijn

lukasmatena, 2020-12-18:
Yeah, we reduced the minimal height again, though not to the original value. I reopened #5162.