prusa3d/PrusaSlicer #5463: Open New Instance on macos has issues

mkoistinen, 2020-12-13:
### Version
2.3.0 beta 3

### Operating system type + version
MacOS Big Sur 11.0.1

### 3D printer brand / version + firmware version (if known)
MK3/S + MMU2/S

### Behavior
In PrusaSlicer preferences, if I set "Allow just single Prusa Slicer instance" (Probably should be "Allow just a single Prusa Slicer instance", notice the preposition "a" added here)

[image]

There exists a Dock menu option to "Open new instance" (in the screenshot it is disabled/gray because there's a modal dialog still open, but normally it is active.)

If I unset "Allow just single Prusa Slicer instance"

Then the option disappears.
[image]

This seems backwards.

Also, even when the Dock menu item "Open new instance" is available, it does nothing.

#### Project File (.3MF) where problem occurs
n/a

lukasmatena, 2020-12-18:
This should be fixed (159dde5) in [2.3.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc1). Could you please retest and close the issue if happy? Thanks.

rtyr, 2021-01-28:
No response. Should be fixed. Closing.

closed 2021-01-28 by rtyr