prusa3d/PrusaSlicer #5464: 2.3 beta3 double-skirt on small objects

sej7278, 2020-12-13:
I noticed that with 2.3b3 on very small models (like inch square) it seems to make two rings of skirt, even though i asked for one. Setting it to zero gives you zero.

The generated gcode then won't open in gcode-viewer and i get this:

```
[2020-12-13 17:39:59.299449] [0x00007fd190ffa5c0] [error]   GCodeProcessor encountered an invalid toolchange, maybe from a custom gcode.
```

If I add another instance of the same object (or scale by 150%) it generates a single skirt and gcode-viewer opens it fine.

Using PrusaSlicer-2.3.0-beta3+linux-x64-GTK3-202012121853.tar.bz2 on Debian Sid.

[demo STL and generated gcode](https://github.com/prusa3d/PrusaSlicer/files/5685102/555bumper.zip)

JohnOCFII, 2020-12-13:
@sej7278 - I've seen that behavior, but I assume it is related to this setting: `Minimum filament extrusion length`:

[image]

sej7278, 2020-12-13:
Maybe, but the skirt is probably 8cm long, not 4mm. Also doesn't explain why gcode viewer won't open it.

foreachthing, 2020-12-13:
It's extruded filament, not length of extruded material through the nozzle 😕 😉

sej7278, 2020-12-13:
Oh yeah doh! I'll have a play

sej7278, 2020-12-13:
yes, changing that from 10 to zero makes it only create a single skirt, however it still segfaults when i double-click the gcode, i don't understand why it will open the gcode from the file menu but not from nautilus, should maybe change the bug title.

sej7278, 2020-12-13:
Added reproducible test case - run `make` to generate the STL and GCODE from the one-liner openscad file (a cube).

Proves it nothing to do with profile/settings/skirts - or even the toolchange warning from the segfault!

[test.zip](https://github.com/prusa3d/PrusaSlicer/files/5685787/test.zip)

enricoturri1966, 2020-12-14:
@sej7278
Could you please share with us the 3mf you used to generate 555bumper.gcode ?
There is something wrong with the extruders count, the GCodeViewer cannot detect the 2nd (left) extruder somehow.

sej7278, 2020-12-14:
printer profile is from @drlex0 https://github.com/DrLex0/FFCP-GCodeSnippets

[3mf](https://github.com/prusa3d/PrusaSlicer/files/5688653/555bumper.zip)

but my test.zip doesn't even use a printer profile and it has the same issue....?

bubnikv, 2020-12-15:
anybody has seen the crash as reported by @sej7278 ?
Maybe it is a hardware or a driver issue?
https://gitlab.freedesktop.org/drm/intel/-/issues/2623

bubnikv, 2020-12-15:
@sej7278 Now I am totally confused. I checked your Makefile, however there is no G-code viewer invocation in there, just slicing.
Does the command line crash?

bubnikv, 2020-12-15:
```
❯ make clean
rm -f test.stl test.gcode
❯ make
/usr/bin/openscad --render -o test.stl test.scad
/mnt/d/release/PrusaSlicer-2.3.0-beta3+linux-x64-GTK3-202012121853.AppImage --gcode test.stl
10 => Processing triangulated mesh
20 => Generating perimeters
70 => Infilling layers
30 => Preparing infill
88 => Generating skirt
90 => Exporting G-code to test.gcode
Slicing result exported to test.gcode
```

Running your Makefile works, no crash experienced.

sej7278, 2020-12-15:
@bubnikv no the Makefle creates the STL from scad, then gcode from the STL

to make gcode-viewer crash you double-click that gcode file

i'm using nouveau not intel

bubnikv, 2020-12-16:
Cannot reproduce on my Ubuntu 20 on WSL2.

Anyways, how did you register prusaslicer to open by double clicking on .gcode?
What desktop are you using and file manager? Where do you click on the gcode file?

sej7278, 2020-12-16:
i don't even need to double-click it, i can just run:

```
/opt/prusaslicer/prusa-gcodeviewer test.gcode 

(prusa-gcodeviewer:729851): GLib-GObject-WARNING **: 10:19:28.776: invalid (NULL) pointer instance

(prusa-gcodeviewer:729851): GLib-GObject-CRITICAL **: 10:19:28.776: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:729851): GLib-GObject-WARNING **: 10:19:28.776: invalid (NULL) pointer instance

(prusa-gcodeviewer:729851): GLib-GObject-CRITICAL **: 10:19:28.776: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:729851): GLib-GObject-WARNING **: 10:19:28.776: invalid (NULL) pointer instance

(prusa-gcodeviewer:729851): GLib-GObject-CRITICAL **: 10:19:28.776: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:729851): GLib-GObject-WARNING **: 10:19:28.776: invalid (NULL) pointer instance

(prusa-gcodeviewer:729851): GLib-GObject-CRITICAL **: 10:19:28.776: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
````

this is debian sid (real install, not wsl) with gnome shell 3.38 and nautilus.

to register it for double-clickability i copied src/platform/unix/PrusaGcodeviewer.desktop to /usr/share/applications (and changed the Exec path).

the key point is that if i use the non-gtk3 build all is fine.

bubnikv, 2020-12-16:
Unfortunately we are out of our wits. We will have to wait for more
feedback.

sej7278, 2020-12-16:
Weird, it's obviously a gtk3 issue but no idea what else I can give you, it's 100% reproducible for me with the gcode files provided.

sej7278, 2020-12-16:
ah, some progress perhaps - just tried it on a pretty much identical debian sid install with beta3-gtk3 but using an intel gpu and it doesn't have the problem - it loads that gcode file fine:

```
$ ~/prusaslicer/bin/prusa-gcodeviewer test.gcode 

(prusa-gcodeviewer:4412): GLib-GObject-WARNING **: 12:28:13.754: invalid (NULL) pointer instance

(prusa-gcodeviewer:4412): GLib-GObject-CRITICAL **: 12:28:13.754: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:4412): GLib-GObject-WARNING **: 12:28:13.754: invalid (NULL) pointer instance

(prusa-gcodeviewer:4412): GLib-GObject-CRITICAL **: 12:28:13.754: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:4412): GLib-GObject-WARNING **: 12:28:13.754: invalid (NULL) pointer instance

(prusa-gcodeviewer:4412): GLib-GObject-CRITICAL **: 12:28:13.754: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:4412): GLib-GObject-WARNING **: 12:28:13.754: invalid (NULL) pointer instance

(prusa-gcodeviewer:4412): GLib-GObject-CRITICAL **: 12:28:13.754: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
```

so perhaps its a nouveau issue?

bubnikv, 2020-12-16:
everything is possible

sej7278, 2020-12-18:
Still getting it with rc1:

```
/opt/prusaslicer/bin/prusa-gcodeviewer test.gcode 

(prusa-gcodeviewer:1037711): GLib-GObject-WARNING **: 00:30:31.893: invalid (NULL) pointer instance

(prusa-gcodeviewer:1037711): GLib-GObject-CRITICAL **: 00:30:31.893: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:1037711): GLib-GObject-WARNING **: 00:30:31.893: invalid (NULL) pointer instance

(prusa-gcodeviewer:1037711): GLib-GObject-CRITICAL **: 00:30:31.893: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:1037711): GLib-GObject-WARNING **: 00:30:31.893: invalid (NULL) pointer instance

(prusa-gcodeviewer:1037711): GLib-GObject-CRITICAL **: 00:30:31.893: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:1037711): GLib-GObject-WARNING **: 00:30:31.893: invalid (NULL) pointer instance

(prusa-gcodeviewer:1037711): GLib-GObject-CRITICAL **: 00:30:31.893: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
Segmentation fault
```

lukasmatena, 2020-12-21:
I'm not sure if this is a duplicate of #5537. Can you try to run the application with `--loglevel=5` to see how far it gets?

sej7278, 2020-12-22:
@lukasmatena doesn't seem to give us much more info, i noticed that it works ok if i don't pass the gcode file as an argument, so probably is same as #5537 (don't suppose there is a nightly/test build with that fix in i can try?)

```
$ /opt/prusaslicer/bin/prusa-gcodeviewer --loglevel=5 test.gcode 
[2020-12-22 14:04:48.584639] [0x00007f15f0a4f5c0] [trace]   System language detected (user locales and such): en_GB
[2020-12-22 14:04:48.585580] [0x00007f15f0a4f5c0] [trace]   Best translation language detected (may be different from user locales): en_GB
[2020-12-22 14:04:48.585609] [0x00007f15f0a4f5c0] [trace]   Switching wxLocales to en_GB

(prusa-gcodeviewer:1906354): GLib-GObject-WARNING **: 14:04:48.695: invalid (NULL) pointer instance

(prusa-gcodeviewer:1906354): GLib-GObject-CRITICAL **: 14:04:48.695: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:1906354): GLib-GObject-WARNING **: 14:04:48.695: invalid (NULL) pointer instance

(prusa-gcodeviewer:1906354): GLib-GObject-CRITICAL **: 14:04:48.695: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:1906354): GLib-GObject-WARNING **: 14:04:48.695: invalid (NULL) pointer instance

(prusa-gcodeviewer:1906354): GLib-GObject-CRITICAL **: 14:04:48.695: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:1906354): GLib-GObject-WARNING **: 14:04:48.695: invalid (NULL) pointer instance

(prusa-gcodeviewer:1906354): GLib-GObject-CRITICAL **: 14:04:48.695: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
[2020-12-22 14:04:48.881585] [0x00007f15f0a4f5c0] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 90MB; Shared memory: 69MB; Private memory: 21MB; Peak memory usage: 90MB
[2020-12-22 14:04:48.882853] [0x00007f15e50a3700] [trace]   Dbus object /com/prusa3d/prusaslicer/InstanceCheck/Object registered. Starting listening for messages.
[2020-12-22 14:04:48.916935] [0x00007f15f0a4f5c0] [info]    Detected gcode producer: PrusaSlicer
[2020-12-22 14:04:48.926717] [0x00007f15f0a4f5c0] [trace]   Loaded G-code generated vertex buffers, 0MB Resident memory: 95MB; Shared memory: 70MB; Private memory: 25MB; Peak memory usage: 95MB
Segmentation fault

```

without gcode argument (using file->open):

```
$ /opt/prusaslicer/bin/prusa-gcodeviewer --loglevel=5 
[2020-12-22 14:05:13.988276] [0x00007f7af27d65c0] [trace]   System language detected (user locales and such): en_GB
[2020-12-22 14:05:13.989204] [0x00007f7af27d65c0] [trace]   Best translation language detected (may be different from user locales): en_GB
[2020-12-22 14:05:13.989233] [0x00007f7af27d65c0] [trace]   Switching wxLocales to en_GB

(prusa-gcodeviewer:1906433): GLib-GObject-WARNING **: 14:05:14.100: invalid (NULL) pointer instance

(prusa-gcodeviewer:1906433): GLib-GObject-CRITICAL **: 14:05:14.100: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:1906433): GLib-GObject-WARNING **: 14:05:14.100: invalid (NULL) pointer instance

(prusa-gcodeviewer:1906433): GLib-GObject-CRITICAL **: 14:05:14.100: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:1906433): GLib-GObject-WARNING **: 14:05:14.101: invalid (NULL) pointer instance

(prusa-gcodeviewer:1906433): GLib-GObject-CRITICAL **: 14:05:14.101: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:1906433): GLib-GObject-WARNING **: 14:05:14.101: invalid (NULL) pointer instance

(prusa-gcodeviewer:1906433): GLib-GObject-CRITICAL **: 14:05:14.101: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
[2020-12-22 14:05:14.290758] [0x00007f7af27d65c0] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 92MB; Shared memory: 69MB; Private memory: 23MB; Peak memory usage: 92MB
[2020-12-22 14:05:14.291997] [0x00007f7add6fe700] [trace]   Dbus object /com/prusa3d/prusaslicer/InstanceCheck/Object registered. Starting listening for messages.
[2020-12-22 14:05:19.252941] [0x00007f7af27d65c0] [info]    Detected gcode producer: PrusaSlicer
[2020-12-22 14:05:19.303399] [0x00007f7af27d65c0] [trace]   Loaded G-code generated vertex buffers, 0MB Resident memory: 119MB; Shared memory: 82MB; Private memory: 38MB; Peak memory usage: 121MB
[2020-12-22 14:05:19.355364] [0x00007f7af27d65c0] [trace]   Loaded G-code generated indices buffers, 0MB Resident memory: 119MB; Shared memory: 82MB; Private memory: 38MB; Peak memory usage: 121MB
[2020-12-22 14:05:19.355709] [0x00007f7af27d65c0] [trace]   Loaded G-code generated extrusion paths, 0MB Resident memory: 120MB; Shared memory: 82MB; Private memory: 38MB; Peak memory usage: 121MB
[2020-12-22 14:05:19.358098] [0x00007f7af27d65c0] [trace]   Refreshed G-code extrusion paths, 0MB Resident memory: 120MB; Shared memory: 82MB; Private memory: 38MB; Peak memory usage: 121MB
[2020-12-22 14:05:25.810846] [0x00007f7af27d65c0] [debug]   message handler shutdown().
```

lukasmatena, 2020-12-22:
@sej7278 Thanks. The log indeed indicates that this is the same as #5537. I will give you a build to test later today if you want.

sej7278, 2020-12-22:
@lukasmatena sure, more than willing to help, just don't think i can compile it

lukasmatena, 2020-12-22:
@sej7278 Can you please try this build? [PrusaSlicer-2.3.0-rc1+28-linux-x64-GTK3-g81878a6](https://drive.google.com/file/d/1fCncKD7P11JDZPJIUNiXp73TSUTE0r1Q/view?usp=sharing)
Thanks.

sej7278, 2020-12-22:
@lukasmatena that builds works flawlessly! :+1: 

```
PrusaSlicer-2.3.0-rc1+28-linux-x64-GTK3-g81878a6-202012221403$ ./prusa-gcodeviewer --loglevel=5 ~/3d_prints/todo/test/test.gcode 
[2020-12-22 15:13:59.434460] [0x00007f2b7cc375c0] [trace]   System language detected (user locales and such): en_GB
[2020-12-22 15:13:59.435734] [0x00007f2b7cc375c0] [trace]   Best translation language detected (may be different from user locales): en_GB
[2020-12-22 15:13:59.435762] [0x00007f2b7cc375c0] [trace]   Switching wxLocales to en_GB

(prusa-gcodeviewer:1932535): GLib-GObject-WARNING **: 15:13:59.551: invalid (NULL) pointer instance

(prusa-gcodeviewer:1932535): GLib-GObject-CRITICAL **: 15:13:59.551: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:1932535): GLib-GObject-WARNING **: 15:13:59.551: invalid (NULL) pointer instance

(prusa-gcodeviewer:1932535): GLib-GObject-CRITICAL **: 15:13:59.551: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:1932535): GLib-GObject-WARNING **: 15:13:59.551: invalid (NULL) pointer instance

(prusa-gcodeviewer:1932535): GLib-GObject-CRITICAL **: 15:13:59.551: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:1932535): GLib-GObject-WARNING **: 15:13:59.551: invalid (NULL) pointer instance

(prusa-gcodeviewer:1932535): GLib-GObject-CRITICAL **: 15:13:59.551: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
[2020-12-22 15:13:59.745125] [0x00007f2b7cc375c0] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 92MB; Shared memory: 69MB; Private memory: 23MB; Peak memory usage: 92MB
[2020-12-22 15:13:59.747013] [0x00007f2b53fff700] [trace]   Dbus object /com/prusa3d/prusaslicer/InstanceCheck/Object registered. Starting listening for messages.
[2020-12-22 15:14:00.038986] [0x00007f2b7cc375c0] [info]    Detected gcode producer: PrusaSlicer
[2020-12-22 15:14:00.139933] [0x00007f2b7cc375c0] [trace]   Loaded G-code generated vertex buffers, 0MB Resident memory: 109MB; Shared memory: 73MB; Private memory: 36MB; Peak memory usage: 109MB
[2020-12-22 15:14:00.166235] [0x00007f2b7cc375c0] [trace]   Loaded G-code generated indices buffers, 0MB Resident memory: 109MB; Shared memory: 73MB; Private memory: 36MB; Peak memory usage: 109MB
[2020-12-22 15:14:00.166407] [0x00007f2b7cc375c0] [trace]   Loaded G-code generated extrusion paths, 0MB Resident memory: 109MB; Shared memory: 73MB; Private memory: 36MB; Peak memory usage: 109MB
[2020-12-22 15:14:00.169445] [0x00007f2b7cc375c0] [trace]   Refreshed G-code extrusion paths, 0MB Resident memory: 109MB; Shared memory: 73MB; Private memory: 36MB; Peak memory usage: 109MB
[2020-12-22 15:14:03.499119] [0x00007f2b7cc375c0] [debug]   message handler shutdown().
```

lukasmatena, 2020-12-22:
Terrific. Closing the issue, thanks for your help. The fix will be in 2.3.0-rc2.

closed 2020-12-22 by lukasmatena