prusa3d/PrusaSlicer #5465: filament_notes parameter being ignored during slicing

tommy11490, 2020-12-13:
### Version
PrusaSlicer Version 2.2.0 +win64

### Operating system type + version
WIN 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3 MMU 2

### Behavior
Hello,

I am on a quest of automating my slicing workflow. Given that I print a lot models that need only one color change always in the same height I have a custom printer profile (custom start, layer change and end gcodes) that simulates wipe tower of MMU2.0. 

Until now, I had 5 printer profiles (every profile load different extruder (T0,T1,T2....)) for each color. Everything worked flawlessly until now.

Now want to make things easier and combine 5 printer profiles to just one. I want the color selection to be done by selecting filament profile.

I wrote small piece of code that utilises filament_notes parameter and assigns T0,T1... commands based on filament profile selected.

What looked like straightforward solution (I already used such a code in the past but with printer_notes parameter on output filename format) seems to not work with filament_notes.

Here is a code I have in a printer profile start g-code:

```
{if filament_notes=~/.*COLOR_PINK.*/}T0
{elsif filament_notes=~/.*COLOR_BLUE.*/}T1
{else}T0
{endif}
```

Here is a code I have in a filament_notes in filament profile:
`COLOR_PINK`

But that does not seems to work. It only works when i change filament_notes to printer_notes and put COLOR_PINK to printer_notes - but that clearly do not fulfill the purpose of my need.

The problematic line of code from gcode debugger in temporary folder is:
```
!!!!! Failed to process the custom G-code template start_gcode
Parsing error at line 30: Referencing a vector variable when scalar is expected
{if filament_notes=~/.*COLOR_PINK.*/}T0
    ^
!!!!! End of an error report for the custom G-code template start_gcode
```

It seems that slicer ignores filament_notes from filament profile during slicing ?

Thank you, best regards.
[image]
[image]
[image]

tommy11490, 2020-12-13:
I got it working by using [0] after filament_notes.
So I indeed was referencing a vector instead of scalar. Strange, because when using printer_notes I do not need to use [0]
```
{if printer_notes=~/.*COLOR_PINK.*/}T0
{elsif printer_notes=~/.*COLOR_BLUE.*/}T1
{else}T0
{endif}
```
Nevertheless, this piece of code works:
```
{if filament_notes[0]=~/.*COLOR_PINK.*/}T0
{elsif filament_notes[0]=~/.*COLOR_BLUE.*/}T1
{else}T0
{endif}
```

bubnikv, 2020-12-14:
because there may be multiple extruders, but just a single printer.
I think it is clear now.
Closing.

closed 2020-12-14 by bubnikv