prusa3d/PrusaSlicer #5469: Featurerequest: Define x inner Perimeter to be printed every second layer to decrease print time

sboehle, 2020-12-14:
### Version
Version 2.3.0-beta3+win64

### Operating system type + version
Windows 10 20h2

### 3D printer brand / version + firmware version (if known)
Anycubic i3 Mega S

### Behavior
I found this feature at IdeaMaker from Raise3D.

There you can define x shells of inner perimeter to be handled as infill.
When you combine these setting with "combine infill every 2 layers" also the inner perimeter shells will be printed every second layer.
For instance you use a layer high of 0.15mm. You combine the infill every second layer.
Than the inner perimeter will be printed with 0.3 mm layer high and the outer perimeter still with 0.15mm.
Resulting in speeding up the print while remaining the quality of the outer shells.

Feature makes sense when you print models with 4 perimeter shells.
Than you can print 2 outer perimeter shells using 0.15mm layer high and two inner perimeter shells with combining every second layer to get 0.3mm layer high. Printing will be speeded up.

Perhaps this feature will be interesting for you. But this feature doesn't make sense for each model, it depends.

#### Project File (.3MF) where problem occurs
n/a

bubnikv, 2020-12-14:
Did Ideamaker implement it? How well does it work.

Frankly I know nobody using the combine infill layers feature.

sboehle, 2020-12-14:
Yes, they implemented it. I haven't tried it. Because i don't use their software. I saw a youtube video with recommendations for speeding up printing without changing something at the speed settings from the printer.

The outcome from the video looked promising for me and i started testing with prusa slicer and combining layers - for infill only at the moment.

Here the link to the youtube video. It is in german, but perhaps there is a translation or you have a colleague speaking german.
https://www.youtube.com/watch?v=e9ahe9NLEUU

Perhaps this combine infill layer feature is underestimated. I never thought about it until i saw that video - now it makes sense to use it - in my opinion.

bubnikv, 2021-01-05:
Ich verstehe Deutsch ziemlich gut.

Starting from the 9th minute, he explains that the approach is only good
for nearly vertical walls. That is also my experience with the combine
infill, which I have personally never used. This seems to me another one
trick pony feature, that will only make the code complex and nobody will
really use.

sboehle, 2021-01-05:
Hi

 

Happy new year 😊

 

Yes. You are right.

 

I use the feature to combine every second infill layer – that is enough for me.

Don’t implement this feature  - it will be resulting in more complex code and therefore more options to “implement” bugs.

I really like PrusaSlicer – in my opinion it is the best in the market.

Cura was the first one I used. Simplyfi3D the second one. They have a very good engine for building support. But the software gets old and I think they aren’t developing the code anymore.

Now with PrusaSlicer I’m very happy.

Also prusament is my favorite one.

I’ll like to buy also an Prusa Printer, but my wife will kill me buying another 3D Printer.

 

Have a nice day

Sylvester

 

po 14. 12. 2020 v 18:36 odesílatel sboehle <notifications@github.com <mailto:notifications@github.com> >

sboehle, 2021-01-05:
Yes you are right. It will make the code more complex. No need to implement it.

closed 2026-09-29 by prusaslicer-bot[bot]