prusa3d/PrusaSlicer #5470: Switching camera views with alphanumeric row on a czech keyboard doesn't work

anymouse404, 2020-12-14:
### Version
2.3.0-beta3, .appimage release.

### Operating system type + version
Ubuntu 20.04

### 3D printer brand / version + firmware version (if known)
unrelated

### Behavior
Switching camera views with alphanumeric row on a czech keyboard doesn't work - using numpad still works.
-With a Czech Keyboard in default view, press shift+1(above the 'Q' key).
-Camera should switch to top view.
-Nothing happens.

Note: This works correctly in 2.2.0 version, and alphanumeric row also works correctly when keyboard layout switched to English.

bubnikv, 2020-12-14:
Works in PrusaSlicer PrusaSlicer-2.3.0-alpha2, broken in PrusaSlicer-2.3.0-alpha3+69
@enricoturri1966 would you please look into it?

bubnikv, 2020-12-14:
I think it broke with ea5fdcd7b1e5e26e105f078ff78dfb4c18bd7743

FidelCapo, 2020-12-16:
Thank you for report. This issue should be fixed in next release.
Closing.

closed 2020-12-16 by FidelCapo