prusa3d/PrusaSlicer #5472: UI bug: profile edit controls hidden by scroll bars

NameOfTheDragon, 2020-12-14:
This is a new UI/usability issue that is not present in alpha 3. It might seem trivial from the description but it is actually a pretty serious and annoying usability issue that potentially creates a lot of friction. Subjectively, I expect It is going to annoy every user every time they use the program, so I think this definitely needs to be considered a 'showstopper'. It should be pretty easy to fix though.

### Version
2.3.0-Beta.3

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
N/A but Creality Ender 3

### Behavior

When the model is sliced, additional information is added to the right hand information panel in the `Info` and `Sliced Info` panels. This causes the panel to grow to a size longer than the window, which causes the vertical scroll bar to appear.
The scroll bar obscures all the the "edit" buttons (gear wheel icon) next to the profile settings drop-downs, so it is difficult to then edit profile settings. No horizontal scroll bar appears, so it is impossible to scroll across to the edit buttons.

The following screen shot shows the condition where the buttons are obscured by the scroll bar.
[image]

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Area5142, 2020-12-14:
Seams the same as #5451

rtyr, 2020-12-14:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5451. Closing.

closed 2020-12-14 by rtyr

lukasmatena, 2020-12-18:
This should be fixed in [2.3.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc1). Let us know if it isn't. Thanks.