prusa3d/PrusaSlicer #5473: Cut tool not accepting keyboard value entry

fasteddy516, 2020-12-14:
### Version
2.3.0-beta3+win64

### Operating system type + version
Edition: Windows 10 Pro
Version: 20H2
OS build: 19042.685

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S (Firmware 3.9.2)

### Behavior
The cut tool does not seem to process 'Z' values entered manually using the keyboard.  Numbers can be entered in the field, but they don't get applied.  Adjusting the height by dragging the graphical control with the mouse works correctly, but is less precise.  (I confirmed keyboard entry works fine in the release version of 2.2, so this appears to have been introduced sometime since then.)

To reproduce, just load any object in the plater view, select the cut tool and try to enter a value manually using the kayboard.

Its-my-right, 2020-12-15:
I noticed this regression too on Beta 3 compared to Beta 2

FidelCapo, 2020-12-15:
Unfortunately, I cannot reproduce this issue. Can you please share video of this issue?

Its-my-right, 2020-12-15:
Hi, I just made this quick video. When it goes from 10 to 36 in the window, I'm actually clicking the border in order to save the value, but it doesn't. Also, I'm trying to press "enter" after typing 10, but it doesn't do anything. In the end you can see I type 10 again, and when clicking "perforum cut", the cut is done at 36mm, as shown in the right pannel.

[PrusaSlicer-2.3.0-beta3 based on Slic3r 2020-12-15 08-41-53.mp4.zip](https://github.com/prusa3d/PrusaSlicer/files/5694077/PrusaSlicer-2.3.0-beta3.based.on.Slic3r.2020-12-15.08-41-53.mp4.zip)

lukasmatena, 2020-12-15:
I can reproduce it in beta3 but not in current master. I believe that @YuSanka has already fixed in with 0f11ff8525d87ebb25c49e5596259b57af8be5e4, is that right?

YuSanka, 2020-12-15:
Yes, this bug was fixed.

closed 2020-12-15 by YuSanka