prusa3d/PrusaSlicer #5474: Legend colors missing on linux since 2.3.0-beta

nordurljosahvida, 2020-12-14:
### Version
2.3.0-beta.1 and later

### Operating system type + version

Ubuntu 20 via non-GTK3 appimage

### 3D printer brand / version + firmware version (if known)

Slicing for MK3S but happens with all printers

### Behavior

Legend is color-less

[image]
[image]

This could have started in the alphas, it's hard to tell, I'll try re-downloading them and see.

sej7278, 2020-12-14:
Have you tried the gtk3 tarball? No real reason to use gtk2 or appimage on Ubuntu 20

nordurljosahvida, 2020-12-15:
Not sure, how would I integrate the tarball into system? I use appimages with appimage-launcher which integrates them with system. Is there a good way without using aliases or custom configs to integrate a simple bin file systemwide?

Anyway, I've tried gtk3 tarball, same issue

FidelCapo, 2020-12-15:
I cannot reproduce this issue with Ubuntu 20 non GTK3..

bubnikv, 2020-12-15:
This looks like something I have seen when fixing HiDPI in OpenGL on Linux / GTK3. #4135 #3245

a13b35ce92efed202ef43f02debb83dfb533d9b0
fixed an ugly rounding issue when populating the ImGUI font map with image thumbnails.

nordurljosahvida, 2020-12-16:
Thanks @bubnikv, I read this is likely going to be the last beta. Should I take the time to compile to make sure this is fixed before stable release? I really hope this doesn't make it into stable, the GTK3 issue might have been there before but this color issue was definitely introduced in beta.1

bubnikv, 2020-12-16:
We are going to release a release candidate hopefully today night.

lukasmatena, 2020-12-18:
@nordurljosahvida
This should be fixed in [2.3.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc1). Could you please retest and close the issue if happy? Thanks.

nordurljosahvida, 2020-12-18:
Perfect thanks, it's solved, both in non-GTK3 appimage and also GTK3 appimage, which now also works with HIDPI

closed 2020-12-18 by nordurljosahvida