prusa3d/PrusaSlicer #5476: 3.3.0 Prusaslicer promised PDF

lmmohr4, 2020-12-14:
3.3.0 beta promised a PDF tutorial. It's now almost 2021!  What's going on, people?

rtyr, 2020-12-14:
?

lmmohr4, 2020-12-14:
NTYR 
What don't you understand?

closed 2020-12-14 by lmmohr4

bubnikv, 2020-12-15:
It is now almost 2021, the current year is 2020. We have the help manual for some time and it is even linked from the application.
https://www.help.prusa3d.com/en/category/prusaslicer_204

We wish you a merry Christmas.

lmmohr4, 2020-12-15:
Thank You, I could not find it. Seasons Greetings Larry Sent from Mail for Windows 10 From: Vojtěch BubníkSent: Tuesday, December 15, 2020 1:20 AMTo: prusa3d/PrusaSlicerCc: lmmohr4; State changeSubject: Re: [prusa3d/PrusaSlicer] 3.3.0 Prusaslicer promised PDF (#5476) It is now almost 2021, the current year is 2020. We have the help manual for some time and it is even linked from the application.https://www.help.prusa3d.com/en/category/prusaslicer_204We wish you a merry Christmas.