prusa3d/PrusaSlicer #5477: Weird translation in Dutch

fcpiek, 2020-12-14:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

fcpiek, 2020-12-14:
When trying out my first hollowing out of  a model, I couldn't find out how to see inside to check the hollow inside. Then I noticed it was maybe due to a weird translation in my Dutch version of Slicer. It says "weergave samenvoegen". In my best translation to english it means "joining displays" which -imho- clearly shows it does not make sense. I would suggest to use "doorsnede weergeven" (display cross-cut).

Tillema, 2020-12-16:
Hello fcpiek,

Thank you for your suggestion. I updated the Dutch translation file (this doesn't go via GitHub). This will be fixed next version of PrusaSlicer that comes with updates on the phrases. From my experience, a translation update is once per 2 or 3 alpha/beta versions, and of course, every full release version.

Regards,
Simon Tillema
Permanent Dutch Translator of PrusaSlicer

The topic can be closed :)

closed 2020-12-16 by bubnikv