prusa3d/PrusaSlicer #5478: Error importing .obk files

bdl3w1s, 2020-12-14:
### Version
2.2.0

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 64 Bit
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
Error importing .obj files. Other obj. files seem to be okay, but this set of files in particular from thingverse doesn't seem to want to import. I had to put into fusion 360 first and then export as an stl.

bubnikv, 2020-12-15:
Neither obj files nor link to the thingiverse projects were provided.

bdl3w1s, 2020-12-15:
Attached files here.
[Visor.zip](https://github.com/prusa3d/PrusaSlicer/files/5696203/Visor.zip)
https://www.thingiverse.com/thing:4661083/files

bubnikv, 2020-12-15:
The obj file contains polygonal faces with more than 4 vertices. We do not support such a file.

bubnikv, 2020-12-15:
For Vojtech's reference:

libslic3r/Format/OBJ.cpp
bool load_obj(const char *path, TriangleMesh *meshptr)
would have to be extended to triangulate polygons. As of now, only quads are triangulated.

libslic3r/Tesselate.cpp,hpp would have to be extended to support triangulation of 3D polygons, possibly returning triangulated triangles by vertex indices, not by vertices.

foreachthing, 2020-12-15:
> The obj file contains polygonal faces with more than 4 vertices. We do not support such a file.

**Ah! That's why!** I've noticed that too, that sometimes a model from Blender will import (as obj) and some don't. So I always convert them to stl (tri-mesh)... Thanks for the hint!

kumavis, 2023-08-23:
encountered this issue -- had to launch prusaslicer from terminal with logs enabled to see error message

lukasmatena, 2026-09-04:
This is kind of on the edge between a feature request and bug report. PrusaSlicer never supported loading of these OBJ files, therefore it is not something that would break now and needed an expedite fix. I consider it a feature request and move it to Discussions to keep our issue tracker free of non-urgent things.