prusa3d/PrusaSlicer #5479: Hungarian language

kakkas, 2020-12-15:
Hey guys!
Im new in github, could you please help me how can i translate Prusa form English to Hungarian language?
What file have to download and translate?

Many thanks,
Oliver

bubnikv, 2020-12-15:
Please see
https://github.com/prusa3d/PrusaSlicer/blob/master/doc/Localization_guide.md

kakkas, 2020-12-15:
> […](#)

Thank you for the help, but i cant open .pot file in the poedit because its said this file is not vaild.
is there any solution?
Best regards,
Oliver

zsolt320i, 2020-12-16:
supporting

bubnikv, 2021-09-22:
Please help creating the dictionaries.
We have no resources to do that ourselves.

closed 2021-09-22 by bubnikv

zomatt, 2022-01-19:
Hello,

Good news is that I have a translation about 80% done.
Some translations are not perfect, some missing, because  I don't find where it is in the software or don't understand the meaning even in English.

Bad news is that the Prusa Slicer handles the cdaracters öÖ, üÜ, őŐ, űŰ incorrectly sometimes.

[image]

What is the way to fix those components, where it fails?

Thanks.

lukasmatena, 2022-01-19:
Could you please point us to the problem on the screenshot so we are certainly talking about the same thing? I assume you mean the question mark in the hint notification at the bottom right. What character is supposed to be there? Can you share the respective line of you PO?
It seems to me that the respective glyph is outside the range loaded by imgui.

zomatt, 2022-01-19:
Yes, it is the tip of the day in the previous screenshot.
Not easy to catch, so here is another place:
[image]

The line is Külső kerület ÁRVÍZTŰRŐ TÜKÖRFÚRÓGÉP árvíztűrő tükörfúrógép, so I may mislead you, it is only űŰ and őŐ.

kocikdav, 2022-01-20:
Yes, we need to set ImGui to use latin-2 encoding for Hungarian (programmatically). I will commit the change. Please @zomatt would you confirm that these are all special letters we need to be showing?
[image]

zomatt, 2022-01-20:
@kocikdav
I can confirm, that would be all.

kocikdav, 2022-01-20:
It is corrected in master.
It would be awesome if we can get your translation into upcoming 2.4.1 release.  Do you think you can make it? You have probably 1 week of time. 
You can make PR with your PO file (even 80%). We will merge it with current state of POT. Then you can continue on updated PO.

zomatt, 2022-01-20:
I'll work on the translation this weekend, so 90-95% is more than possible by Monday.
Then I can send the files to you through which channel?

kocikdav, 2022-01-21:
[Here](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/DK/PrusaSlicer-2.4.1-alpha0+32-win64-g6fed5c29a-202201210000.zip) is the latest build where all letters should be fine and also contains the latest POT file.
There are two options to send us the translation. First and best is to make fork of our repository, commit your changes, make pull request and we will merge it. You also get contributor badge here on github.
But if you are not familiar with git, it is possible to send us your work via email. My is [here](mailto:david.kocik@prusa3d.cz).

zomatt, 2022-01-25:
Hello @kocikdav,

I used a little outdated pot file, but I hope, it is not a big problem.
Anyways, the translation has been sent by email, please check.