prusa3d/PrusaSlicer #5481: legend forgets "Show stealth mode" after each rendering

3Dietrich, 2020-12-15:
### Version
2.3.0-beta3+

### Operating system type + version
macOs 10.15.7

### 3D printer brand / version + firmware version (if known)
i3Mk3S
### Behavior
#### _the problem_ 
Legend is set to "Show normal mode". When I want to show it in stealth mode I need to switch it every time when it got sliced
#### _Steps needed to reproduce the problem_
- show legend: it's set to [Normal mode]
- switch it to [Stealth mode]
- change something so that it needs to re-slice, if not automatically: call re-slice
- look again at the legend.. it's back to "Normal" mode
#### _Expected Results_
- it should remain on showing the last chosen mode, even it's "Stealth" mode
#### _Actual Results_
it switches allways back to Normal mode

#### Project File (.3MF) where problem occurs

lukasmatena, 2020-12-18:
This should be fixed (52e0e5b) in [2.3.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc1). Could you please retest and close the issue if happy? Thanks.

3Dietrich, 2020-12-20:
jawoll!
danke!

closed 2020-12-20 by 3Dietrich