prusa3d/PrusaSlicer #5482: Triangles (and stars) are not centered

andrewdefilippis, 2020-12-15:
### Version
PrusaSlicer-2.2.0+-202003211132

### Operating system type + version
macOS 11.0.1 Big Sur

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Triangles are not centered, and it causes the Octagram Spiral to be offset.

[image]

[image]

[image]

#### Project File (.3MF) where problem occurs
N/A

neophyl, 2020-12-15:
Its centred on the square bounding box of the object.  Multi part objects also are treated as one bounding box for the purposes of any type of infill.
I think I've seen an issue already with a request to be able to set the centre point of infills

andrewdefilippis, 2026-09-02:
Adding a comment to have this revisited or resolved if fixed.